OpenHD/OpenHD #325: Image Resolution and HD screen

thelordash, 2020-05-24:
Hello I am trying to build a handheld version of OpenHD sadly at the moment the only rasberypi screen small enough but with high enough resolution is the wave share 5.5inch

and the screen comes in portrait mode to switch it t landscape you use display_rotate which uses quite a bit of GPU the result is really bad latency  2-5sec. and I wish there was a setting to reduce the resolution of the image from the camera to match the pi limited resources while retaining the quality of that display.

thanks

steveatinfincia, 2020-05-31:
I think we talked about this in telegram when you posted this here, but for the sake of having a record the thing we will do is handle rotation in the new osd, which will always work and will not affect latency or CPU use.

jcausemann, 2020-07-03:
I have tried landscape with this 5" display:
LS055R1SX03 5,5 zoll 2k IPS LCD modul 2560*1440 
https://de.aliexpress.com/item/32815697328.html

With no Succsess. Seems to be very intricate or perhaps with settings in config.txt not possible.
https://www.raspberrypi.org/forums/viewtopic.php?t=224229

If you do it with new OSD, that is a different approach?
regards Jochen

RespawnDespair, 2020-12-04:
There is an [issue ](https://github.com/OpenHD/QOpenHD/issues/126)for this in the QOpen.HD repository. 

I am closing this issue to prevent double efforts.

closed 2020-12-04 by RespawnDespair