OpenHD/OpenHD #329: Antenna configuration

mj11232, 2020-05-27:
Why not recommend two antennas on the airpi, but on the ground pi? Is there interference between two or more antennas for sending telemetry or uplink signals?

steveatinfincia, 2020-05-31:
As I recall there wasn't much to gain from having 2 wifi cads on the air side, at least if they're both using identical omnidirectional antennas. However some wifi cards do have 2 antennas and you can use those on the air side.

RespawnDespair, 2020-12-04:
It is certainly encouraged to have two antennas on the air side if your WiFi adapter supports it. It is generally discouraged to have two (or more) WiFi adapters in the air unit as it adds almost no benefits but does increase the weight.

So the recommendation is not so much regarding 'antennas' as it is about actual WiFi adapters. The documentation should reflect this now.

closed 2020-12-04 by RespawnDespair