OpenHD/OpenHD #331: Sending custom data via telemetry

user456813, 2020-05-30:
I have a arduino board collecting some data in my drone, how can I send these data to the GroundPi and log it?
sorry for my bad English.

steveatinfincia, 2020-05-31:
What sort of data are you collecting? We will probably have an official way to do this sort of thing so people don't have to hack solutions together.

tsrab, 2020-06-05:
Hi, 

I have the same idea. 
Collecting data on a Arduino board or attach sensors to GPIO pins on the raspberry. I'm  planning to use it on a firefighting robot. 
Data i want to send to the grond station:
Engine status
Engine errors
Engine temperature
Hydaulic errors
Hydraulic temperature
GPS location
Fuel level
Environment temperature

Is it possible wire a GPS receiver to the GPIO pins of the air PI and send the data to the ground PI?

Greetings Bas van den Berg

steveatinfincia, 2020-06-07:
That sounds like something ArduRover should do rather than the pi to avoid reinventing the whee. All of it would be much easier including the GPS and sensors if you were using a cheap controller board like a matek with ArduRover on it.

With that set up it would be very easy to send the data you're talking about.

tsrab, 2020-07-10:
I have some success.
Using https://github.com/danzimmerman/barebones_MAVLink as a base.
I can send gps and accelerometer data at this time.

RespawnDespair, 2020-12-04:
@tsrab that sounds like the correct route.
Adding the messages to MAVLink makes sure we adhere to the open standards used so far.

The project you referenced seems like a good starting point for people that want to achieve the same.

I will close this issue as the proposed solution appears to work. We might link people to this issue when the subject comes up in the future. I might also add some description to the documentation. Did you by chance expand on your initial attempt? Anything we could use in writing some documentation?

(En met zo'n naam moet je wel Nederlands zijn?)

closed 2020-12-04 by RespawnDespair

tsrab, 2020-12-04:
Ja inderdaad nederlands.👌
Momenteel werkt de arduino voor mavlink nog niet stabiel. Moet er nog wat
tijd in steken om het goed te krijgen.

Als ik wat heb dat goed werkt zal ik het posten.

Grt Bas

RespawnDespair, 2020-12-04:
If you have any updates, please post them, we can add it to the documentation. 

Bedankt!