OpenHD/OpenHD #335: power wifi rtl8812au

kasatka60, 2020-06-06:
openhd 2.0.0rc6
"Taobao Card" rtl8812
[image]
[image]

# >>>>>>>>>>[TXPOWER-AIR/GROUND]-- For Atheros AR9271 or RTL8812/14AU Cards ONLY
TxPowerAir=58
TxPowerGround=58

# >>>>>>>>>>[TXPOWER-RALINK]-- (Not confirmed Working]
TXPOWER=2000

which parameter affects the power?
maximum allowable power?

steveatinfincia, 2020-06-07:
The TX power settings are a little weird at the moment, I can test it tonight and see what it's doing.

steveatinfincia, 2020-07-02:
Checked out the code, the TXPOWER=2000 setting is correct for realtek cards, I see the same thing being done in the driver documentation. 

So our settings file is technically wrong. The unit is mBm, so 2000mBm = 20dBm.

The range seems to be up to 3000, but I don't think increasing it will necessarily be any better.

closed 2020-07-03 by kasatka60