OpenHD/OpenHD #338: VEYE/Innomaker Cameras overshooting bitrate / causing runaway latency

careyer, 2020-06-14:
**Describe the bug**
VEYE/Innomaker cameras are found to overshoot the "set/desired bitrate" limit causing severe runaway latency issues in no time: 

E.g:
(1) measured bitrate is 13Mbit/sec... (
(2) set bitrate is 5.2Mbit/sec (with bitrate-percentage=40 in openhd-settings-1.txt) 
(3) regardless of (2) the actual videobitrate will hit 9.5-10Mbit/sec easily when moving the camera and will only slowly calm down again.

**To Reproduce**
Steps to reproduce the behavior:
1. use VEYE/Innomaker IMX290 (likely other VEYE cameras models are affected the same)
2. limit the bitrate via openhd-settings-1.txt file considerably to like 40%
3. start the system
4. move the camera arround so that scenary changes
5. observe the actual video bitrate exeeding acceptable limits by a huge margin

Regarding 4. I was able to observe that bitrate is likely to shoot up with specific scenaries more than with others. I could hardly reproduce the effect in my living room (much details, rather dark)... but in my bedroom (less detail but much white colors and bright) the effect would be massively exegarated.
Maybe has something tto do with internal sharpening algorithm or something alike.

**Expected behavior**
actual bitrate should not ovberhshot set/desired bitrate by too much. 

**Configuration and debug logs**
n/a. Can share my openhd-settings-1.txt if needed 

**Hardware:**
 - Air side
 -- Pi3B
 -- Innomaker VEYE IMX290
 -- Alpha 036NHA

 - Ground side
 -- Pi3B+
 -- 2x TP-Link 722N, 1x Ubiquiti Wifi-Station EXT
 -- 7" Genuine PiTouch Display 

 - RC/telemetry uplink
 -- RC over Open.HD
-- GeeekPi Joystick controller

Thank you very much for taking care of that issue! 👍 
Besz regards
Thomas

steveatinfincia, 2020-06-28:
Oddly enough, when i try this on buster I can make it happen easily, but on stretch I can't. Same scene and lighting all the time.

Makes me think there is something in the pi firmware we'll have to track down and ask them for help with.

raphaelscholle, 2023-01-31:
Runnaway latency is looked after in 2.3

closed 2023-01-31 by raphaelscholle