OpenHD/OpenHD #339: Display rotate  cause extremely high latency

ansmax, 2020-06-19:
Hi, 
   When I use a wave share 5.5inch as my OpenHD monitor,I must set "display_rotate=3"in config.txt and it will cause extremely high latency(about 5 sec)  same like [issues#325](https://github.com/OpenHD/Open.HD/issues/325)
   First I think it's monitor problems,but when I try another monitor and set  "display_rotate=X", the latency becoming very high.
  is this a bug or hardware limitation?

pilotnbr1, 2020-06-20:
Try using extra param settings in the openhd-settings-1.txt file (or set in the qopenhd app)
-hf
-vf

https://github.com/OpenHD/Open.HD/wiki/Expert-~-Raspivid-camera-settings

ansmax, 2020-06-20:
> Try using extra param settings in the openhd-settings-1.txt file (or set in the qopenhd app)
> -hf
> -vf
> 
> https://github.com/OpenHD/Open.HD/wiki/Expert-~-Raspivid-camera-settings

I think its flip but not rorate

steveatinfincia, 2020-06-20:
Rotating the screen that way uses the pi GPU processor, which only works for low resolutions since it's not handled by accelerated hardware. As a result the GPU can't keep up with the job and falls behind quickly.

What we'll have to do is [rotate in QOpenHD](https://github.com/OpenHD/QOpenHD/issues/126), which will work properly.

closed 2020-06-22 by ansmax

codev-cory, 2022-01-17:
-rot, --rotation : （0,90,180,270 deg）