OpenHD/OpenHD #340: Mavlink telemetry do not work in INAV

ansmax, 2020-06-22:
Hi,
I met a problem when I try OPENHD in INAV.
If I set the telemetry to LTM and the band is 115200(both INAV port setting and OPENHD setting),it can work(half),the attitude shows fine but there is no battery info no GPS info no speed no distance info,all information show 0 except attitude.
Then I try to set telemetry to Mavlink and the band is also 115200(both INAV port setting and OPENHD setting),nothing works.

so whats my problem?

FC:MATEKF722MINI
AIRPI:3A
GNDPI:3B+
VER:2.0.0 RC7 [Stretch]

Another question,does these hardware can run Buster img or it only support pi4？

closed 2020-06-24 by ansmax

steveatinfincia, 2020-06-27:
Hey did you get this working or do you still need help? :)

> Another question,does these hardware can run Buster img or it only support pi4？

Yep buster runs on everything

cgrrty, 2021-02-09:
I need help 
I set the telemetry to LTM or MAVLINK are not can work 
FC:F4 V2 PRO
AIRPI:3A
GNDPI:3B+
VER:2.0.0 RC7 [Stretch]