OpenHD/OpenHD #344: Open.HD-2.0.0rc7-buster

domenic64, 2020-06-26:
Open.HD-2.0.0rc7-buster:
GND PI3A+ , AWUS036NEH RT2870/3070 RT2800USB , RPCAM 2.1
It doesn't start.

[image]
[image]

On Open.HD-2.0.0rc7-stretch **it works**.

steveatinfincia, 2020-06-27:
Hey :)

Hmm, what does the lower panel on the boot screen say when it does this on buster?

domenic64, 2020-06-28:
personal settings
KEYFRAMERATE=5
RC=disabled
TELEMETRY_UPLINK=disabled
VIDEO_TMP=sdcard
VIDEO_SAVE_FORMAT=mp4   (is the file saved in raw, no mp4)
RF usb RT3070 drv RT2800USB
RPCAM 2.1

1) Tests rc09-buster GND pi4B 4G ok it works 
2) Tests rc09-stretch GND Pi3A+   ok it works
3) Tests rc09-stretch AIR rp0  ok it works 
I have the impression that there is less CPU load of previous versions maybe because I set up KEYFRAMERATE=5?
is it possible to reboot on air after changing the settings? my air no motors no rc

4) Tests rc09-buster GND Pi3A+ NO WORK
it blocks
[image]
[image]
[image]

Tests rc09-stretch GND Pi3A+  ok it works
[image]
rssi usual problems

thank you

steveatinfincia, 2020-06-28:
Ok, I'm going to order a 3A+ so I can test it, sorry for the trouble :)

steveatinfincia, 2020-06-28:
> is it possible to reboot on air after changing the settings? 

There is a remote power control system yes, it just doesn't have a button yet, will get that added this week.

TheNewJavaman, 2020-08-25:
Is this still an issue? 2.0.0 release is now out.

steveatinfincia, 2020-08-25:
I've tested it now, 3a+ works ok on the air side with a workaround applied, but there is still a kernel/firmware issue in Buster that affects using the 3a+ as a ground station.

Stretch on 3a+ as a ground station should be working at the moment.

the-nicolas, 2020-12-01:
Any update here? Is using stretch for ground still the workaround?

RespawnDespair, 2020-12-04:
Several bugfixes and an update to the kernel were added over the last few releases. The latest release 2.0.6 is downloadable now.
I don't have a Raspberry Pi 3A+ for testing, so i would like to ask you to verify if the problem still persists in the latest version. If you are unable to test, please let me know, i will find another way to validate.

As a side note, the Raspberry Pi 3A+ is not supported as a GROUND unit officially. (see https://openhd.gitbook.io/open-hd/hardware/sbcs).

This is mainly due to extra hardware demands for the groundstation that the 3A+ (like the Zero) just can't deliver consistently.
But please see if it will work for you, just know it isn't officially supported.

Eagerly awaiting your feedback.

domenic64, 2020-12-04:
sorry, Saturday and Sunday I can test v.2.06

RespawnDespair, 2020-12-04:
That would be perfect!

domenic64, 2020-12-05:
GND PI3A+ , AWUS036NEH RT2870/3070 RT2800USB
open hd 2.06 strech ok work .
Buster not work
1 boot
[image]
2 boot
[image]
3 boot
[image]
3 boot after the boot screen black screen
pi3A led green & led red turned on fixed
[image]

domenic64, 2020-12-05:
Is there a log file on the usd can i figure out what happens? Tell me where I find it that I make a copy.

steveatinfincia, 2020-12-07:
Logs won't show anything for this one, from what I've seen something runs out of memory (not anything we're running ourselves, it's likely one of the kernel/firmware memory allocations). It may go away entirely with 5.4 or 5.10 kernels, I'll check it ASAP.

closed 2022-10-16 by raphaelscholle