OpenHD/OpenHD #346: Open.HD-2.0.0rc11-stretch

domenic64, 2020-07-07:
No engine,No RC
settings:
KEYFRAMERATE=5
RC=disabled
TELEMETRY_UPLINK=disabled
VIDEO_TMP=sdcard
GND PI3A+  AWUS036NEH RT3070 RT2800USB 
AIR , PI0       AWUS036NEH RT3070 RT2800USB, RPCAM 2.1
it works ok.  (rssi -126 usual problems)

GND PI3A+ CINESE RTL8812BU  DRIVER rtl88x2bu ID 0bda:b812  not works, hw not recognized
[image]

It is possible by adding on AIR an rs232 gps to send the data to GND and QOPENHD.
Thank you

steveatinfincia, 2020-07-07:
Hey :)

RTL8812BU support is being added, may require some additional changes to make it work well.

TheNewJavaman, 2020-08-25:
You may want to re-title this issue.

steveatinfincia, 2020-08-25:
This basically looks like a request for rtl88x2bu WiFi card support, we have a driver for it but it needs some changes to work properly.

RespawnDespair, 2020-12-04:
@steveatinfincia was support implemented recently? I remember the 8188, but not sure about this.

@domenic64 As for your second question 

> It is possible by adding on AIR an rs232 gps to send the data to GND and QOPENHD.

Can you please create a new issue for that with some additional description of the situation and what you want to achieve?

domenic64, 2020-12-04:
Yes
thanks

domenic64, 2020-12-04:
CINESE RTL8812BU driver rtl88x2bu ID 0bda:b812
Ok work with v 2.06

RespawnDespair, 2020-12-04:
It appears the issue was resolved, I have seen the new issue for the GPS. 

Closing this one.

closed 2020-12-04 by RespawnDespair