OpenHD/OpenHD #348: [Feature request] AirPi position to GroundPi on OSD

ABLomas, 2020-07-09:
On route from analog to digital - trying to get things small as possible and integrating all into HMD.
With analog it's not so hard to fly with up to ~10dBi patch on head, turning with semi-long range plane or quad together. Analog has instant feedback about signal quality so it's easy to "tune" to plane position.
With digital it's different story, RSSI display is not instant, visual signal lags, so directional antenna with only visual feedback is hard to use.

So, idea - need to integrate AirPi direction to OSD. No need for full tracking or coords calculation - we can use existing tracker for that, just instead of driving servos - it would need to turn OSD arrow. For simplicity only direction needed, altitude data could be discarded (tbh i don't think arrow is needed, angle to airplane is enough (180...-180°) so even less work).
Now - how to push back data? If i understand correctly - we have no spare ports (spare taken by tracker, which would calculate position). Sure, it's possible to add mag to RPi itself and calculate direction from here, but i think, it's not video transmission system job. Softserial? What data format to take (so it could be easy to parse)?

P.S. before anyone offers to build tracker - i have one. But having tracker - longer setup, need to carry tripod, then wires to HMD, etc etc etc - need something portable enough and still semi-long range.

ABLomas, 2020-07-10:
Well, i think inter-board communications will take more work to implement, compared to tracking on Ground-RPI side with mag-only.
What about this:

- connect mag to ground pi (I2C)
- calculating heading from mag is pretty much trivial task (calibration and declination is not so trivial, but doable and we are not limited with few kB of memory like on 8-bit controllers)
 - home position = ARMing position (do not know if it's possible to fetch them from mavlink, maybe check channel values and on arm store current coordinates)
 - get aircraft coordinates, calculate heading vs HMD heading, get difference
 - display difference on OSD (now here is a problem, have no idea how new OSD works, maybe it's hard to implement new element - on old OSD displaying numeric value was not hard task)

Do i miss something?

pilotnbr1, 2020-07-10:
I guess this all assuming the viewing device is not a phone. If it's a smart phone then phone only needs to share orientation (mag/gyros)

If only a screen attached to pi then yeah you need to pump mag in from a sensor. 

Qt widget is not hard. Just different. Hardest part is probably setting up project. Stephen made an example widget. Just follow that template

ABLomas, 2020-07-10:
I can make code for reading/calibrating mag (including auto mag. declination) - not sure what programming language is preffered (have perl daemon from some other project, but not so hard to rewrite in some other language), but integrating everything will be hard task. Need help on this one.
Phone as display device - good idea, but doubt that many ppl are using phones for long range flight (FPV goggles with HDMI/HMD or ext. screens, even on TX are much more popular IMO). Mag is 2$ PCB on 4 wires - trivial enhancement from HW side...

RespawnDespair, 2020-12-04:
The idea is sound and it looks like the solution is already sketched out. I hope we can add this feature in 2.1, @ABLomas indicated he needs help doing that, so it will have to be put on the backlog for now.

So not closing this, but also no promises on when we will have time to work on this.

Consti10, 2023-02-01:
please open an issue on QOpenHD if you still wanna press forward with this

closed 2023-02-01 by Consti10