OpenHD/OpenHD #349: QOHD RSSI widget gets stuck

CopterGUI, 2020-07-13:
When starting a OHD system the new RSSI widget in RC11 of the QOHD OSD gets stuck.

This video shows a screen recording of iOS QOHD 0.4.0b8-dirty while the Airside is booting up and connects to the GS.
Its especially relevant because it shows the one and only moment the widget seems to work fine - shortly after the telemetry stream arrives - but before the video starts.

https://youtu.be/910csegGSrY

The same behavior can be seen on a connected HDMI monitor.

I tested all possible combinations of stretch/buster in air and groundPi but its always the same issue.

Rebooting ONLY the AirPi is resetting the widget for some seconds, again only untill the video starts.
Rebooting ONLY the GroundPi does not change the stuck value.
Rebooting ONLY the the r9mm receiver does not change the stuck value.
Rebooting ONLY the Flight controller does not change the stuck value.

I think this is a software problem, I exchanged all the hardware part by part but dont see any difference.

The live RSSI values can be monitored in Mission Planner connected via USB to the FC due the whole process, its always fine working.

The old OSD system is working fine with my hardware and arduplane settings.
I tried different SRx values and baud rates. still no luck.
 
Steps to reproduce the behavior:
1. Boot up Groundstation running RC11
2. Turn on Radio (Taranis Qx7 with r9m module in my case) in range test mode 
3. constantly connect / disconnect the antenna to trigger a wide movement of rssi
4. Boot up Airside running RC11
5. Wait till telemetry stream starts and OSD shows values. RSSI widget is working good now.
6. Few seconds later when the video stream starts the RSSI widget gets stuck at the last value.
All other widgets remain working.

**Expected behavior**
RSSI widget should get updated like the other widgets.

**Configuration and debug logs**
datarate 3
40fps
rest is at default

**Hardware:**
 - Air side
 -- Pi model: Pi3a+
 -- Camera(s): tested with: PiHQcam / GoPro
 -- Wifi card(s): tested with 722n / 036nha
 -- Other hardware: Matek405-ctr running Arduplane

 - Ground side
 -- Pi model: Tested with: Pi4-8gb Pi4-2gb Pi3b+
 -- Wifi card(s):722n / 036nha / generic atheros
 -- Hotspots: WIFI
 - RC/telemetry uplink
 -- Do you use RC over Open.HD, or a separate rf link?: External R9M long range module. R9mm receiver.

**Client devices (in case of hotspot/tethering issue)**
 - Device: iPhone8
 - OS: recent
 - App: QOHD
 - Version: 0.4.0b8
 - Connection method: Wifi Hotspot

I saw this is working for norbert, but im all out of ideas what could be wrong.

steveatinfincia, 2020-07-13:
Very weird, definitely a software problem though. I'll check it out again today.

jcausemann, 2020-07-13:
Hi
Your video:
https://www.youtube.com/watch?v=910csegGSrY&feature=youtu.be
is marked as private.
:-(