OpenHD/OpenHD #350: [Feature request] Navio2 support

LimitlessGreen, 2020-07-15:
It would be great, if Emlid's Navio2 could be supportet in OpenHD, because a Pi is needed anyway.
I'm not sure, if it should be included in the "normal" image, maybe a second one with Navio support.

There is an image builder for stretch and buster (https://github.com/emlid/pi-gen-navio)
Emlid is providing deb packages and since our build system is now package based too, it shouldn't be too much work.

Important parts I filtered out:
- Repo infos, keys, ...
https://github.com/emlid/pi-gen-navio/tree/navio-buster/stage3_emlid/02-apt-tweaks/files
```
deb http://files.emlid.com/navio2/debian stretch/buster main non-free
```
- RCIO kernel module packages for communication with the controller board
https://github.com/emlid/pi-gen-navio/tree/navio-buster/stage3_emlid/04-rcio
https://github.com/emlid/rcio-dkms
```
rcio-dkms
rcio-firmware
```
- Ardupilot packages (for some reasons no 4.x for copter in repo, maybe need to be compiled manually :/)
```
$ apt search ardupilot
Sorting... Done
Full Text Search... Done

ardupilot-meta-3.6/unknown,now 3.6.11-2.0 armhf [installed,automatic]
ardupilot-navio-copter-3.6/unknown,now 3.6.11-1.0 armhf [installed,automatic]
ardupilot-navio-plane-4.0/unknown,now 4.0.0-1.0 armhf [installed,automatic]
ardupilot-navio-rover-3.4/unknown 3.4.2-1.0 armhf
ardupilot-navio-rover-4.0/unknown,now 4.0.0-1.0 armhf [installed,automatic]
ardupilot-navio-sub-3.5/unknown,now 3.5.4-1.0 armhf [installed,automatic]
ardupilot-navio2-copter-3.6/unknown,now 3.6.11-1.0 armhf [installed,automatic]
ardupilot-navio2-plane-4.0/unknown,now 4.0.0-1.0 armhf [installed,automatic]
ardupilot-navio2-rover-3.4/unknown 3.4.2-1.0 armhf
ardupilot-navio2-rover-4.0/unknown,now 4.0.0-1.0 armhf [installed,automatic]
ardupilot-navio2-sub-3.5/unknown,now 3.5.4-1.0 armhf [installed,automatic]
ardupilot-service-3.6/unknown,now 3.6.11-1.0 armhf [installed,automatic]
```

- ROS is also in the default images for advanced robot purposes, but I don't think this should be relevant for us.

---

Maybe I have to extend this list.
I tried to build an image locally, but for some reasons my local build machine is always failing.
```
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 openhd : Depends: openhd-microservice (= 0.1.13) but 0.1.14 is to be installed
          Depends: veye-raspberrypi (= 20200704.1) but 20200706.1 is to be installed
E: Unable to correct problems, you have held broken packages.
```
[buildlog.log](https://github.com/OpenHD/Open.HD/files/4926749/buildlog.log)

steveatinfincia, 2020-07-19:
The package thing should be resolved now, I was hoping apt would let us pin specific dependencies to version them together but it simply doesn't work that way, it tries to install newer packages and then breaks when it tries to configure them and notices the `= 0.1.13` constraint. I moved them all to `>=` now.

LimitlessGreen, 2020-07-19:
Thank you very much for fixing it :+1:

What do you think is the best practice for the `rcio` kernel module? Pack it with the kernel and do not load it by default or have it separated in an own DEB package?

RespawnDespair, 2020-12-04:
@LimitlessGreen , this is still very relevant I think, more people have asked for this.
It is probably however 'the other way around' from the current Open.HD images. 
What we are doing there needs to be added to a custom fork of the emlid builder. (I think that's what you are doing actually)
It might be prudent to wait for 2.1 to become available since it has a cleaner architecture and might integrate better with the emlid code.

I will put this issue on the todo for 2.1, but feel free to keep using it to share information.

closed 2023-01-31 by raphaelscholle