OpenHD/OpenHD #352: is there have frame drawing

dreamy2019, 2020-08-01:
I'm interested in doing more research,is there have frame drawing for my quick start.
sorry for my bad English. Thanks for a great god.

steveatinfincia, 2020-08-13:
Hey, frame drawing like a diagram of the system? There's one on the main page of the [wiki](https://github.com/OpenHD/Open.HD/wiki).

dreamy2019, 2020-08-14:
谢谢无私的帮助和奉献，请教您都是怎么调试openhd项目代码的？

steveatinfincia, 2020-08-15:
> 谢谢无私的帮助和奉献，请教您都是怎么调试openhd项目代码的？

在目标设备上运行代码。有些事情也可以在PC上运行。

Run the code on the target device. Some things can also run on the PC.

RespawnDespair, 2020-12-04:
现在，相关项目中只有自述文件。 在不久的将来（很有可能在2.1版的公共发行版中），我们还将尝试提供更详细的开发文档。
由于问题似乎已得到解决，因此我将关闭此问题。
如果您还有其他问题，请随时打开一个新问题。

Right now there are only the readme files in the relevant projects. In the near future (most likely with the public release of version 2.1) we will also try to provide a more detailed development documentation.
I will close this issue as the question seems to be answered. 
Feel free to open a new issue if you have another question.

closed 2020-12-04 by RespawnDespair