OpenHD/OpenHD #353: 2.0.0rc11 building wifi drivers

mcfly13, 2020-08-02:
I have bunch of wifi cards which is not supported currently

`0bda:f179 Realtek Semiconductor Corp`

So i tried to build a driver against the current build which is

`4.19.122-v7+`

Got all prep for kernel source (checkout raspberrypi kernel on specific commit)

 download the rtl8188eus driver from aircrack-ng
https://github.com/aircrack-ng/rtl8188eus

but I'm having error on `make`

```
  Building modules, stage 2.
  MODPOST 1 modules
FATAL: parse error in symbol dump file
make[2]: *** [scripts/Makefile.modpost:92: __modpost] Error 1
make[1]: *** [Makefile:1526: modules] Error 2
make[1]: Leaving directory '/usr/src/linux'
make: *** [Makefile:2246: modules] Error 2
```

Same error when I tried different drivers

I've done the same steps on the ezwifibroadcast build for rtl8812au which works.

Is it an issue for this build? or Am I doing it wrong? Is there a guide how to build wifi drivers on the buster/stretch based builds?

steveatinfincia, 2020-08-03:
If injection actually works I’ll get it added this week, there are more things that have to change than just the driver

mcaras, 2020-08-03:
Thanks @steveatinfincia 

btw, is there a guide for it to allow other wifi devices to be detected by openhd after the driver is built?

mcflyph, 2020-08-04:
Thanks @steveatinfincia 

btw, is there a guide on how to do the other procedure aside from building the wifi driver?

I'd like to help on testing other wifi devices that has monitor and injection mode

steveatinfincia, 2020-08-13:
The rtl8188eus driver is in there now, however mine has not arrived yet so I have not tested it, let me know if it works.

mcaras, 2020-08-15:
Thank you @steveatinfincia! will test in now

mcflyph, 2020-08-15:
thanks @steveatinfincia will test in now!

steveatinfincia, 2020-08-15:
Set the frequency manually, rc17 still has 8188 marked as a 5ghz card so the auto setting won't work.

mcflyph, 2020-08-15:
Got it.

IGNNE, 2020-09-05:
Closed my issue, my bad for not reading through the existing ones.

RespawnDespair, 2020-12-04:
It seems the original issue was fixed with the added support for the 8188 cards. I won't close it right away as I'd like confirmation from @mcflyph or @IGNNE that the cards work. (Don't have one available to check myself)

mcflyph, 2020-12-04:
@RespawnDespair 
I will start testing again to confirm.

 Thanks!

mcflyph, 2021-02-26:
Nope, still doesn't work on `0bda:f179 Realtek Semiconductor Corp`

raphaelscholle, 2022-03-27:
Since I'm currently working on the kernels, I just ordered two cards, which (should) contain that chipset, if that's the case I can test some things. 
I know those drivers are inside the kernel-builder, but needed fixing for the newer kernels, ... that fix should be done, but isn't tested until I have some cards

closed 2022-10-17 by raphaelscholle