OpenHD/OpenHD #358: [Feature Request] telemetry from bluetooth

whoim2, 2020-08-08:
Hello! Ground have telemetry on gpio, its good. But this need external bt module to connect telemetry to android device, example.
May be, use internal rpi bt module for send telemetry to android?

steveatinfincia, 2020-08-13:
Yea we've had a few people ask about it, I'll add it to the new telemetry system going in for 2.1.

In the meantime you can actually connect an esp32 to the hotspot and get telemetry, people have been doing that for antenna trackers to make them wireless. Look around the [openhd forum](https://discuss.openhdfpv.com/) for details.

whoim2, 2020-08-16:
> Yea we've had a few people ask about it, I'll add it to the new telemetry system going in for 2.1.
>
> In the meantime you can actually connect an esp32 to the hotspot and get telemetry, people have [...]

Thanks! I may use 2x paired BT module, e.g. hc-05/06 or JDY-30/31.
Wait for 2.1.

Consti10, 2023-02-06:
Not planned r.n. use uart out on gnd.

closed 2023-02-06 by Consti10