OpenHD/OpenHD #359: No OSD data Ground Pi from mini pix

seba1213, 2020-08-09:
No OSD data is provided after launching the Ground Pi, only the flight mode is displayed. But if I connect to mavlink via mission planner, all data is displayed until I not reboot the flight controller.

I am using 2.0.0rc15 Stretch, Pi 3a +, Pi Zero, mini pix radiolink, fatshark hd2
I tried to configure Stream Rates Setup from this instruction (https://github.com/OpenHD/Open.HD/wiki/RC-~-RC-with-Ardupilot#stream-rates-setup), I also tried to change the protocol to mavlink2 it did not help ...

steveatinfincia, 2020-08-10:
Turn on QOpenHD in the settings file and look at the status area in the settings panel (connect a mouse temporarily, click the openhd logo), does it show telemetry heartbeats arriving regularly at the bottom?

seba1213, 2020-08-10:
I can't do this on a graund pi because the pi3a only has 1 usb and wifi is soldered to it. I checked on mobile app qOpenHD Android, the problem on it is also there - displays the flight mode and messages like PreArm:need 3d fix...
Last openhd heartbeat 29-218
Last telemetry heartbeat 22 - 833
I noticed that connecting QGroundControl also solves this problem.

steveatinfincia, 2020-08-10:
Ah I know what’s going on, the flight controller doesn’t have the telemetry messages turned on, QGC enables them temporarily.

Look in the OpenHD wiki, there’s a page about telemetry that shows which parameters to enable in mission planner, they’re named like SRx_EXTRA, x being the number of the UART you’re using to connect to the air pi.

closed 2020-12-03 by steveatinfincia