OpenHD/OpenHD #360: ENABLE_OPENHDVID=Y gives no video bandwidth, black video feed

TheNewJavaman, 2020-08-11:
**Describe the bug**
Enabling the new feature gives 0.0mbit and a black video feed. OSD and telemetry work fine.

**To Reproduce**
Steps to reproduce the behavior:
1. Clean image of RC15
2. Change ENABLE_OPENHDVID=Y
3. Boot up

**Expected behavior**
Normal video feed, just using the new element

**Configuration and debug logs**
only one change, as listed above

**Hardware:**
 - Air side
 -- Pi model 3a
 -- Camera(s) pi v2 with arducam lens
 -- Wifi card(s) 36nha
 -- Other hardware
 - Ground side
 -- Pi model 3b+
 -- Wifi card(s) 36nha
 -- Ground Control Station(s) qopenhd android b10
 -- Hotspots - do you connect you GCS using ethernet, wifi and/or usb tethering: wifi
 - RC/telemetry uplink ibus
 -- Do you use RC over Open.HD, or a separate rf link? rc over openhd

**Client devices (in case of hotspot/tethering issue)**
 - Device: [e.g. iPhone6] galaxy s9
 - OS: [e.g. iOS8.1] android 10
 - App - FPV_VR, QGC, and version qopenhd b10
 - Version [e.g. 22] b10
 - Connection method: wifi

**Additional context**
Add any other context about the problem here.

Everything else works fine. Just the video stops working. Black screen, 0.0mbit connection.

steveatinfincia, 2020-08-11:
This is likely a bug I fixed a few hours ago, one of the arguments to the openhdvid program (the recording bitrate setting) was conflicting with another one, should be resolved today.

TheNewJavaman, 2020-08-11:
rc 16 or 2.0? i'm excited

steveatinfincia, 2020-08-13:
This is released now and should be working, let us know if you have any more trouble.

https://github.com/OpenHD/Open.HD/releases/tag/2.0.0rc16

TheNewJavaman, 2020-08-15:
It's still there in rc17

steveatinfincia, 2020-08-15:
Ok I'll test it today

steveatinfincia, 2020-08-15:
Fixed for real this time, there was a typo :)

Tested it just now and all looks ok.

closed 2020-08-15 by steveatinfincia

steveatinfincia, 2020-08-23:
> Still there in 2.0.0. Now it's directly tied to the AIR_VIDEO_RECORDING settings. openhdvid works, but air-side recording does not.

Ok