OpenHD/OpenHD #362: Using Open.HD with AR9285 based mini PCIe card

milanerd, 2020-08-12:
Hello everyone,

It is impressive to see what people have managed to achieve using this and other WFB based repos. Congratulation guys!

Maybe it's not the best place to ask this question, but I am trying to run the Open.HD (as well as EZ-wfb and svpcom's wifibroadcast) on a Gateworks Ventana board with Ubuntu 18.04 using MikroTik R11e-2HPnD PCIe card that uses Atheros AR9285 chip. Both aerial and ground side have the same hardware, the air side is connected to Pixhawk with APM. I am sending a UDP video stream at 3 Mbps together with 2-way UDP telemetry.
In my opinion this is quite a powerful setup, when I set up the gnd side as a WiFi AP and the air side connects to it, I can achieve about 1.5-2 km communication range in rural areas. With Open.HD and other wifibroadcast forks I would like to achieve longer range if possible.

Currently I can build Open.HD and run it, and while the sender and receiver are close to each other the video runs nicely with a few gray frames every few seconds. However the transmission range with this setup is really short, maybe about 20-30 meters, when the video goes completely gray (snow).
I have noticed that the video gets much better when we approach the drone and hold it. I thought that it is maybe due to some shadowing or signal reflections, but I've read recently that it is actually an antenna diversity issue.

I did play with different bitrates settings using iw, I tried both legacy and ht-mcs rates, I also tried various combinations of mcs index, stbc and ldpc but it didn't really improve the range. (In some of the repos I have read that stbc and ldpc should be 0 and that atheros cards don't do well with ht-mcs, but even with legacy rates it's the same).
I have also to fix the data rate to minstrel_ht, but I didn't notice any improvement.

I have read that probably ath9k doesn't apply the bitrate settings provided by iw. What I didn't do, and what I probably should do is to patch the ath9k driver in my kernel, however I have never done this before. I have found that DroneBridge and RespawnDespair have the kernel patches in their repos, but those are patches for kernel 4.14, I am using 5.4 currently.

In your opinion, would patching the ath9k and rebuilding the kernel solve the issue (or at least be a good step towards a solution)?
Could you maybe point me to someone who already did or trying to do something similar?

My apologies if this is not the right place to ask this and thank you in advance for your help.

Best regards,
Milan

steveatinfincia, 2020-08-12:
Part of this might be differences in the way that wifi chip works, and there are also significant changes in some kernel versions that make it difficult to just upgrade. 

We're going to release x86 images in a couple weeks so we might revisit this at that point.

The ventana is kinda big, what kind of drone are you putting it on?

steveatinfincia, 2020-08-13:
Ah the Ventana isn't x86 is it, hmm. Still likely to be differences in the chip.

milanerd, 2020-08-13:
Thank you for your fast response!

I have a correction, the chip used is AR9582, not AR9285.

Probably there are chip differences, but I still hope that with your help I can figure something out and that wfb can be done with this chip as well.

Yes, Ventana boards isn't x86, its ARM. Actually its size isn't really an issue at all, I am using the small one Ventana 5510 (35x70mm) and it's on a custom quadcopter.
https://www.gateworks.com/products/imx6-single-board-computer-gateworks-ventana-family/gw5510-single-board-computer/

I will try to manually apply the ath9k patches I found on other repos (hoping that chipsets would work in the same way), rebuild the kernel and try to see if that solves the issue.

RespawnDespair, 2020-12-04:
There has been some interest in the Gateworks boards before.
One of the goals of the 2.1 version is to support other SBC's. With some help from you the Gateworks could be one of those boards.
As for the AR9582, we need to double check that is capable of the correct modes (injection and monitor).

For now I'll add it as a todo to the 2.1 project.

raphaelscholle, 2023-01-31:
should be possible in x86 2.3

closed 2023-01-31 by raphaelscholle