OpenHD/OpenHD #364: AirPi disconnected when use 2 wifi adapter

ansmax, 2020-08-20:
**Describe the bug**
When I use 1 ASUS AC56 adapter on GroundPi , Image transmission and RC control work good . but when I use 2 ASUS AC56 adapter on GroundPi ,most of the time they will work together for 2-3 sec and disconnected suddenly，sometimes one of them cannot work( adapter LED off ) .
I tried every 2.0.0 RC version of OPENHD and both Pi3B+ and Pi4B , the problem still occurs.
I once considered additional USB hub , the problem still occurs.
I also replaced the adapter , when I installed adapter A ,it work fine , when I installed adapter B ,it work fine also, but when I installed both adapter A and adapter B  the problem still occurs.
OPENHD Configuration is default .
Looking forward to your reply.

steveatinfincia, 2020-08-20:
Are they soldered?

ansmax, 2020-08-20:
NO , I use USB plug

steveatinfincia, 2020-08-20:
Are there power wires soldered to the USB pads on the underside of the pi?

If no power wires are soldered anywhere then that is going to be the cause, the pi USB ports can barely handle 1 ac56 card, I suspect 2x causes them to lose power and reset which is what you’re seeing.

The usb connectors on those cards are also prone to disconnect because of vibration/tiny movements. It would be nice to have a better solution than telling people to solder wires, but at the moment that’s the best option.

ansmax, 2020-08-20:
> Are there power wires soldered to the USB pads on the underside of the pi? If no power wires are soldered anywhere then that is going to be the cause, the pi USB ports can barely handle 1 ac56 card, [...]

thanks a lot, I well try.

ansmax, 2020-08-21:
> Are there power wires soldered to the USB pads on the underside of the pi? If no power wires are soldered anywhere then that is going to be the cause, the pi USB ports can barely handle 1 ac56 card, [...]

It is solved！
2 POINTS:
Wires may not be soldered but DP+ and DP- must twined together.
USB can not provide 2 AC56 adapter.

closed 2020-08-21 by ansmax