OpenHD/OpenHD #367: Is it possible to get device list

FeedTheWorst, 2020-08-26:
hi, i'm running latest release on a pi3b+ (gnd) and pi3(air), I have a decent video quality, a good telemetry feedback, but it's like my controller is not connected, I use a custom one made from an arduino pro micro (device is like a joystick, it's almost like the geekpi one from the builds section  -> https://github.com/OpenHD/Open.HD/wiki/Hardware-~-DIY-builds ), driver should natively be compatible, so i wonder if it's just badly detected or if it's a software related issue, i also use ibus for the rc protocol, which is send to a arduino nano to convert it to ppm (because the fc don't have enought uart and a serialrx and telemetry cn't share the same one), the custom "joystick" is correctly detected in windows as well as linux (raspbian).

steveatinfincia, 2020-08-26:
This is likely a bug that was fixed the other day and will be resolved in 2.0.1, sorry for the trouble :)

FeedTheWorst, 2020-08-27:
ok thanks for reply

closed 2020-12-03 by steveatinfincia