OpenHD/OpenHD #368: [Feature Request] TBS Crossfire Mavlink over wifi support

jgilbertfpv, 2020-08-29:
TBS crossfire TX modules can output mavlink data via the built in wifi module. I came across a post in the forum for this but have not been able to test it because i'm on a capped data plan with basically dial-up connection speeds so i haven't been able to download everything to be able to try to compile and test myself https://discuss.openhdfpv.com/t/injecting-mavlink-telemetry-from-udp-outside-i-e-crossfire/160

jgilbertfpv, 2020-08-30:
used sudo nano to add the changes to try this out today, i was only able to get flight mode to display (using Inav) but i may not have everything configured just right.

wx4cb, 2020-09-03:
i replied to that post (quoted here) as I couldn't get it to work either, The way the crossfire mavlink/wifi works is that once you have connected to the AP or connected IT to an AP, you listen for the UDP broadcast on whatever port you have setup in the crossfire module, then just reply with a mavlink heartbeat to that and it will start sending you data on that port.

> silly question, but why don’t you (seeing as you’re already using a pi with 2.4ghz wifi built in) just connect to the crossfire AP using that then open a connection to the module by looking for its UDP broadcast.
> 
> That is actually the better way to do it. as that’s how the wifi bridge in the crossfire was designed to be used. connect to the AP, listen for the mavlink heartbeat, reply to the heartbeat, get the data.
> 
> I can’t see how you are “automatically” getting the mavlink data from the crossfire in there without connecting to the UDP port of the crossfire ?

jgilbertfpv, 2020-09-04:
I'll need to study up on the connections a bit more. i think i understand what you're getting at though.

fingadar, 2020-09-04:
Look at https://discuss.openhdfpv.com/t/injecting-mavlink-telemetry-from-udp-outside-i-e-crossfire/160/6

wx4cb, 2020-09-10:
> Look at https://discuss.openhdfpv.com/t/injecting-mavlink-telemetry-from-udp-outside-i-e-crossfire/160/6

couldn't get that going @fingadar - right now i'm just trying with a simple hc05 on a usb serial port plugged into a ground pi, an dthe setup works fine on mission planner and MWPtools for inav, but i get no data on the ground pi. i've tried setting debug=Ybut having trouble trying to find a console

RespawnDespair, 2020-12-04:
This might be related to #373 
I'm not sure it is, but that issue makes me suspect there is at least an issue with external data.

Could you give an update on the current status of this issue?

wx4cb, 2020-12-07:
@RespawnDespair it's still an issue - i would guess it's a similar issue to the other one, but i've given up on openHD for now for what I'm doing  as due to the size of the airframe (sub250) having a flight controller in there as well as a rPI-0 is just too much space for now.

If the other issue and this can get fixed then it's possible that it will work as is. and i'll be a happy camper again :D

Almischalmi, 2020-12-07:
Just to save you some trouble and since it looks like you don’t it yet: Mavlink through the CF module works right now only by connecting the Wifi Module to an AP, not by connecting TO the CF module.

wx4cb, 2020-12-07:
> Just to save you some trouble and since it looks like you don’t it yet: Mavlink through the CF module works right now only by connecting the Wifi Module to an AP, not by connecting TO the CF module.

oh i know. i'm on the alpha team for tbs and i already use it perfectly fine with mission planner and my tracker.

steveatinfincia, 2021-03-21:
This is mostly implemented, we’ll still need to get the telemetry service to actually connect to whatever port is open on the crossfire, maybe just a simple option to connect to a specific up would work

wx4cb, 2021-03-21:
> This is mostly implemented, we’ll still need to get the telemetry service to actually connect to whatever port is open on the crossfire, maybe just a simple option to connect to a specific up would work

correct, all you have to do is look for the UDP broadcast from the module and then hit that port (usually 8888) and it will start sending data.

closed 2023-01-31 by raphaelscholle