OpenHD/OpenHD #369: Add support for TPLink TL-WN722N V3.20

IGNNE, 2020-09-01:
I have successfully established a connection with a TPLink TL-WN722N **V3.20** using the original [befinitiv/wifibroadcast](https://github.com/befinitiv/wifibroadcast).

In order to use this chip in monitor mode, I used the driver from [aircrack-ng/rtl8188eus](https://github.com/aircrack-ng/rtl8188eus), which compiled fine on a Raspberry 3A+ with the latest Raspberry OS.

To replace the driver and switch the device to monitor mode, I used the following commands (wlan0 is the adaptor):
```
modprobe -r r8188eu
modprobe 8188eu
ifconfig wlan0 down 
iw dev wlan0 set type monitor
ifconfig wlan0 up
iwconfig wlan0 channel 1
```

After that, I was able to `rx`/ `tx`a camera video stream from the Raspberry. Performance wasn't that good, but I believe my parameters were far from optimal. Also, both devices were sitting less than 1m away from my regular access point, and there are several other apartments around me.

I would like to try and configure the adaptor with Open.HD, but I don't know how to continue from here. Since the adaptor is not officially supported and needs a special driver, I believe I have to edit some parts in Open.HD, but where can I start? 

Thanks in advance for your answers.

steveatinfincia, 2020-09-01:
The driver is already in there, it just needs some changes in order to correctly support some kernel networking parameters, for some reason they were left out of that driver. I should have time to look at it again this weekend.

IGNNE, 2020-09-02:
Great to hear that. If I can help with testing or anything, let me know. :+1:

IGNNE, 2020-09-05:
Since this is more or less a duplicate of #353, I'm going to close it

closed 2020-09-05 by IGNNE