OpenHD/OpenHD #371: Question: OpenHD on x86 Hardware (Ground Side)

wx4cb, 2020-09-08:
Currently playing with OpenHD on a pair of pi 3b+s , is it possible to run OpenHD on an x86 based machine - (image or not, not a big deal, but preferable for a script version).

Reason why I ask, my Current GCS system for INAV is running ubuntu and mwp tools etc so I already have a linux install on that machine.

I would like to repurpose a spare linux box that I have for OpenHD as it has a far more powerful hardware footprint than a pi and would also allow me to run multiple things on the same machine.

pilotnbr1, 2020-09-12:
its being worked on but not currently available now

wx4cb, 2020-09-13:
@pilotnbr1 if you want any help luke, let me know.

AMoll1, 2020-10-16:
Yes, x86 support that would be great

IGNNE, 2020-10-28:
I played a bit with gst-trace and the vaapi gst plugin on my Intel Whiskey Lake to see if there are any decoder problems, since I could not find good answers online. The vaapi decoder seems to be too fast to be real, I'm at >>1 ms for the decoder. I guess it somehow breaks the measurement?

Anyway, the whole pipeline including (cpu) encoding and (gpu) decoding takes around 10 ms (without displaying), which is quite good imho.

Here's the full command I used:
`GST_SHARK_LOCATION=profiler GST_DEBUG="GST_TRACER:7" GST_TRACERS="interlatency" gst-launch-1.0 videotestsrc is-live=true ! 'video/x-raw,width=1920,height=1080,framerate=30/1' ! x264enc threads=8 tune=zerolatency speed-preset=ultrafast  ! vaapih264dec low-latency=true  ! xvimagesink`

RespawnDespair, 2020-12-04:
This looks like a very interesting discussion, I'm going over all the issues to make sure they are still up to date and assign them to the right projects. As stated in several other issues (but not know at the time of this issue) support for X86 is something we have on the roadmap for 2.1. That's why I added this issue to that project and labeled it 'enhancement' (couldn't find a better label).

It is also an open invite to help develop and test the 2.1 version, hopefully we can share the code for 2.1 soon so we can start co-developing the system with interested users.

steveatinfincia, 2020-12-07:
For the most part there won't be a need for any significant changes to support x86 hardware in OpenHD 2.1. There are a handful of little things in the new `openhd-system` service (which is in git right now in the master branch) that figures out what kind of board the system is running on, but aside from that everything works like a normal linux system, if there are usb cameras connected they'll get detected and used just like arm boards. 

The exception to that will be x86 platforms that may have camera connectors (CSI/DVP/HDMI perhaps), but I haven't seen one of those yet.

wx4cb, 2020-12-07:
that's awesome... i can't wait to help test some sauce 👍

IGNNE, 2020-12-08:
There are some Intel boards that use CSI cameras, for example my old, cheap Intel tablet. But I don't think those are really that important, and the Linux support is not good, anyway.

I'm more interested in the found side, because just bringing a laptop would simplify my setup (it already has a screen, a huge battery, a hdmi port...). Thanks for the work here! :+1:

iamfaith, 2021-01-29:
Hello all, are there any prebuilt ground side docker images that can be run on the X86 machine now?

wx4cb, 2021-01-29:
@iamfaith why a docker image? Just curious... my case is to use something like my nuc sized ground station

iamfaith, 2021-01-29:
> @iamfaith why a docker image? Just curious... my case is to use something like my nuc sized ground station

The scenario for me is that I can pick up my laptop as a ground side while I go outside for an FPV flying. Image if you are using Pi as the ground side you still need a portable screen as well as the charger. So why not use a laptop with stronger performance?

then docker helps everyone set up this environment easily. Just my thought.

wx4cb, 2021-01-29:
> The scenario for me is that I can pick up my laptop as a ground side while I go outside for an FPV flying. Image if you are using Pi as the ground side you still need a portable screen as well as [...]

Obviously if you're using a raspberry pi, then this thread is moot :D but yes if i'm going to be out for a while, i will take my whole built up ground station. but 9/10 times i'm by myself so generally will use my laptop.

Not all of us are very well versed in docker (me included) but adding a layer of complexity such as that imho, would cause more headaches than they're worth. As opposed to having a whole debian package / redhat RPM would be preferred.

That way it would pull all the dependencies in and you're then not stuck with trying to maintain a whole set of dependencies in an environment when things change

CkovMk, 2021-06-16:
Wow, Really looking forward to this. I already have a x86 SBC on my UAV running ROS. In order to use OpenHD, I added another Rpi3b. If I could intergrate OpenHD directly to SBC, I would be able to get rid of that Rpi. Also, GCS config could be simplified, too.

MorrisR2, 2022-01-26:
> That way it would pull all the dependencies in and you're then not stuck with trying to maintain a whole set of dependencies in an environment when things change

That's precisely the problem Docker _solves._ You mentioned "whole debian package / redhat RPM would be preferred".
Debian is the seventh most popular Linux distro, based on Distrowatch stats.  So that approach would make packages that do not work for the top six most popular distros.

A Docker container works for _any_ distro.  Any version of any distro.
It also provides a a base amd64 package that you can do toher things with.  If you'd rather spend days messing with librariy dependencies and backporting the code the earlier libraries rather than 10 minutes installing Docker, you could do that.  it would give you somewhere to start from. You could modify as needed to run without Docker for one particular version of one particular distro.

CorruptName, 2022-06-22:
This would be more useful now than ever with raspberry pies being in short supply.

raphaelscholle, 2022-10-17:
2.2.x will have a x86 version, but it is linux only

closed 2022-10-17 by raphaelscholle