OpenHD/OpenHD #372: about the wifi card and HDMI to CSI2 card

chenhugi, 2020-09-13:
hello everyone 
     After several days of experiment, I found several problems. About WiFi card
When I want to enable dual NIC settings, only one NIC is working.
I bought an HDMI to csi2 adapter card. The chip is Toshiba's tc358743. When I used this card, I found that the prompt driver was incompatible. Moreover, when I connected the BMD camera, the picture color was not correct. I don't why .

steveatinfincia, 2020-09-13:
Dual NIC where, on the ground?

HDMI-CSI board should work fine already, lots of people use it. Is your camera sending 1080p60? It can’t support that frame rate at 1080p, might try 1080p30 or 720p60 if you can.

chenhugi, 2020-09-13:
I used two wifi card on air and groud . For csi board , i think the issues is not the fps ,maybe it is the Color range .

RespawnDespair, 2020-12-04:
Hi, sorry for not coming back to this earlier.
Have you made any progress since your last post?

Using two compatible WiFi cards on air and ground should require no additional setup or config alterations.
Could you explain you use case a little more?

As for the HDMI to CSI board, as long as the board itself is working the problem most likely will be with the camera feeding the signal into the board. They are usually limited to certain resolutions and fps, anything beyond that might produce unpredictable results.

I will leave this issue open while waiting for feedback, maybe then we can determine if there is an actual issue with Open.HD or if it's settings related.

jpmarotta, 2022-02-11:
@RespawnDespair 

I have a BMPCC4K available, will grab an additional wifi card and test this weekend.

closed 2022-10-16 by raphaelscholle