OpenHD/OpenHD #373: question: External mavlink to ground side

wx4cb, 2020-09-14:
i currently have mavlink data going to my groundstation via bluetooth from the crossfire module (HC-05 with FTDI Uart)

This works perfectly fine with mission planner and qground control etc so i know the link is working. 

My issue is that i have the exact same dongle plugged into the ground side rPi and i have set "external" for mavlink and set the baud rate and device node in the configuration. but i'm not getting any mavlink data into the osd (pitch roll etc).

If I plug the module into the air side rPi and set them to wbc with the airside pi being "external" - exact same config as the ground side, it works fine.

All I want is the mavlink data coming into the ground side rPi via a uart instead of the wbc link. any help appreciated

Here is my current ground side config: https://pastebin.com/r3wufT1a

wx4cb, 2020-09-14:
reproduced the ground side rPi section here:

```
# >>>>>>>>>>[TELEMETRY WBC/OTHER]--
# wbc = use wifibroadcast as telemetry up/downlink
# external = use external means as telemetry up/downlink (LRS or 3DR dongles)
# if set to external, set serialport to which LRS or 3DR dongle is connected
# both on ground and air pi
TELEMETRY_TRANSMISSION=external
 
# >>>>>>>>>>[TELEMETRY UPLINK]--
# Set to "disabled" or "mavlink" for Mavlink (QGC, Missionplanner, etc.)
TELEMETRY_UPLINK=mavlink
 
# >>>>>>>>>>[USB OPENHD-RC SERIAL PORT/BAUDRATE]--Serial Port Settings between Air Pi & Flight Controller.
# Serial port and baudrate (19200 is minimum) 
# Set this to "/dev/serial0" for Pi onboard UART serial port or "/dev/ttyUSB0" for FTDI adapter. 
FC_RC_SERIALPORT=/dev/ttyUSB0
FC_RC_BAUDRATE=57600
 
# >>>>>>>>>>[OPENHD-FC SERIAL PORT/BAUDRATE]-- Air Pi and Flight Controller 
# *Should be the same as FC_RC_BAUDRATE 
FC_TELEMETRY_SERIALPORT=/dev/ttyUSB0
FC_TELEMETRY_BAUDRATE=57600
 
# >>>>>>>>>>[EXTERNAL DEVICE SERIAL PORT BAUDRATE]--Serial port settings if using TELEMETRY_TRANSMISSION=external
EXTERNAL_TELEMETRY_SERIALPORT_GROUND=/dev/ttyUSB0
EXTERNAL_TELEMETRY_SERIALPORT_GROUND_BAUDRATE=57600
 
# >>>>>>>>>>[GROUND SERIAL PORT OUT ENABLE?]--Set to "Y" to enable output of telemetry to serialport on ground Pi
# (for antenna tracker etc.)
ENABLE_SERIAL_TELEMETRY_OUTPUT=N
 
# >>>>>>>>>>[GROUND SERIAL PORT/BAUDRATE OUT]-- 
# Used for ground Pi telemetry output if TELEMETRY_OUTPUT=Y
TELEMETRY_OUTPUT_SERIALPORT_GROUND=/dev/serial0
TELEMETRY_OUTPUT_SERIALPORT_GROUND_BAUDRATE=57600
```

wx4cb, 2020-09-15:
note.... according to the wiki, this can be done:

https://github.com/.../Software-~-Advanced-~-Telemetry-&-OSD

`If you have some other means of transmitting the telemetry to the ground (e.g. an LRS with serial downlink or 3DR dongles) choose TELEMETRY_TRANSMISSION=external to receive the telemetry stream on the ground Pi serial port. If using external telemetry transmission, also configure EXTERNAL_TELEMETRY_SERIALPORT_GROUND= and EXTERNAL_TELEMETRY_SERIALPORT_GROUND_BAUDRATE=`

RespawnDespair, 2020-12-04:
Thanks for the detailed report, and yes, it was definitely possible to do this, as documented in the Wiki.
I suspect this has broken over the last releases as this is not commonly used by most members and thus has gone unnoticed until now.

I will add it to the stability backlog.

wx4cb, 2021-02-26:
has this been looked into yet? *nudge nudge wink wink*

steveatinfincia, 2021-03-21:
Yes this is implemented, you can connect whatever you want to the uart or usb serial on each side and the messages will reach the OSD

wx4cb, 2021-04-20:
> Yes this is implemented, you can connect whatever you want to the uart or usb serial on each side and the messages will reach the OSD

I still can't get this working (even on a pi4)  is there specific things I need to set in the config file that aren't in the wiki? 

I know the mavlink is working fine as the same mavlink stream is also running mission planner

Consti10, 2023-02-01:
2.3-evo supports and requires mavlink over wifibroadcast, you can still use an external dongle for transmission as a backup if you want to, but not to get data into QOpenHD.

closed 2023-02-01 by Consti10