OpenHD/OpenHD #374: Bug? Can't get PI4 to boot

wx4cb, 2020-09-14:
I tried loading the latest release but it won't boot on the pi4. It appears that the dtbs are there, but there's no entry for the pi4 in the config.txt file.

Previous issue: https://github.com/OpenHD/Open.HD/issues/185

```
[pi0]
start_file=zero_x.elf
fixup_file=zeroup_x.dat
[pi0w]
start_file=zero_x.elf
fixup_file=zeroup_x.dat

[pi1]
start_file=1to3b_x.elf
fixup_file=1to3bup_x.dat
[pi2]
start_file=1to3b_x.elf
fixup_file=1to3bup_x.dat
[pi3]
start_file=1to3b_x.elf
fixup_file=1to3bup_x.dat
[pi3+]
start_file=start_x.elf
fixup_file=fixup_x.dat
```

 bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Configuration and debug logs**
Please attach the relevant openhd-settings*1.txt and any debugging logs available

**Hardware:**
 - Air side
 -- Pi model
 -- Camera(s)
 -- Wifi card(s)
 -- Other hardware
 - Ground side
 -- Pi model
 -- Wifi card(s)
 -- Ground Control Station(s)
 -- Hotspots - do you connect you GCS using ethernet, wifi and/or usb tethering
 - RC/telemetry uplink
 -- Do you use RC over Open.HD, or a separate rf link?

**Client devices (in case of hotspot/tethering issue)**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - App - FPV_VR, QGC, and version
 - Version [e.g. 22]
 - Connection method

**Additional context**
Add any other context about the problem here.

steveatinfincia, 2020-09-14:
The firmware override lines aren't actually needed, we should really just remove them all.

In any case I just created a new 2.0.1-buster image and pi4 started right up, what is the exact model of your pi4? Is it the 8gb model?

I think there's someone else who has an 8gb though, quite a few people would be complaining if pi4 didn't work at all.

wx4cb, 2020-09-14:
yes it's the 8gb model. but i had a little issue, so i reflashed with a new one and it seems to boot up ok. (i get the osd :D) 

now to figure out the screen resolution, the touch screen response and getting the mavlink data in from a usb serial dongle

closed 2020-09-14 by wx4cb