OpenHD/OpenHD #375: OpenHD don't work when 4G Dongle is pluged

ShenMiaozhang, 2020-09-22:
OpenHD don't work when 4G Dongle is pluged.

The version I used is Open.HD-2.0.1-buster.img

[image]
[image]

ShenMiaozhang, 2020-09-22:
https://discuss.openhdfpv.com/t/open-hd-4g-lte-problem/216/3

pilotnbr1, 2020-09-22:
If you do not see your LTE stick connecting to ZeroTier its probably a problem with the install on the AirPi. You can delete ZeroTier tracker txt file created in boot partion of the Airpi image (this will force a reinstall attempt) and hook up a monitor to Airpi to see whats going on. The ZeroTier tracker file gets created upon every install attempt. If that fails you can look at an OpenHD image or in the github repo within wifibroadcast scripts and the file lte_functions.sh. Within lte_functions.sh are the install steps- try those manually on a pi with your lte stick and attempt a manual installation

RespawnDespair, 2020-12-04:
The new docs [describe ](https://openhd.gitbook.io/open-hd/advanced-setup/lte-connection#setup-lte)the process of setting up LTE, the also mention the need to have a compatible WiFi card connected as well. 

This functionality is not really supported mainstream yet, if you get it working that's great, all the information we have should be in the docs. 

Feel free to submit any new issues you find (or ask questions in the Discuss forums), for now i will consider this issue closed.

closed 2020-12-04 by RespawnDespair