OpenHD/OpenHD #376: New osd artifical horisont trouble

schs2, 2020-10-05:
I have trouble with new OSD in Open.HD-2.0.1-buster.img 
When aircraft rotated - the horisont ladder rotated with jumps, not smooth.
Stretch image with old OSD provide smooth movement.
Hardware - PI 3b+.
Setting - default. 
Monitor connected via HDMI.

Example new osd
https://youtu.be/6u2ljdhWcJI

Example old osd
https://youtu.be/QxOCND4XG8E

RespawnDespair, 2020-12-04:
Both videos are set to private, so I can't see what's going on, but it sounds like the Stream Rates issue we have documented. Are you sure the Stream Rates are set the same in both tests? (see the [docs](https://app.gitbook.com/@openhd/s/open-hd/software-setup/telemetry-and-osd/~/settings/share#stream-rates-settings))
If the Flight Controller settings are not different between the tests we will definitely have to look into this.

schs2, 2020-12-04:
I change youtube video permissions. It's not stream rate issue. When I look via old OSD - horisont is OK, when I look via qopenhd on pi3b+ horisont is twitch.
When I look via phone(wifi hotspot) - all ok.

RespawnDespair, 2020-12-04:
I think the old OSD has some interpolation going on that 'lerps' the actual movement making it appear smoother. 
Also the old OSD was done in low level OpenGL which is faster and also adds to the smoothness.

As luck would have it, the horizon on the new OSD has undergone a rewrite as of late.

@pilotnbr1 Is this something that has been improved in the new horizon widget?

pilotnbr1, 2020-12-04:
The refactored horizon should be more efficient in qpaint/c vs the qml one. Will be interested to see if there is any visible improvement in the next qopenhd version..

RespawnDespair, 2020-12-04:
I hope we can release it soon, meanwhile I'll leave this ticket open and attached to the 2.0 stability project.

schs2, 2020-12-04:
Thanks. I change groud PI to pi4b, it got a little better. Android qopenhd work wery well.

schs2, 2020-12-25:
After change groud PI to pi4b problem resolved.

closed 2020-12-25 by schs2

pilotnbr1, 2020-12-25:
2.0.7 onwards the horizon (and flight path vector) were totally rewritten in C vs the old qml for the "new" osd
This should help cpu performance.