OpenHD/OpenHD #378: Alternative handdling of mavlink

olliw42, 2020-10-12:
Hey
this is just about sharing with you an other way of including your own mavlink messages which you may find easier to deal with and maintain. Currently, you have to generate an extra fork of mavlink and include it as submodule. A - IMHO - simpler procedure is to only submodule the standard mavlink repo, and use a python script which calls the code generator. This allows you to place the own dialect xml anywhere in a folder you want to, and avoids having to mess around with that extra repo. Updating to the latest mavlink repo then only needs updating the submodule sha. I'm using this since a while, and found it really to be so much easier, hence this post. You can see that e.g here: https://github.com/olliw42/opentx/tree/owpr-mavtelem-v11/radio/src/thirdparty/Mavlink. 
Keep up with your good work, and have fun
Olli

steveatinfincia, 2020-12-03:
Cool, sounds like a good idea thanks!

RespawnDespair, 2020-12-04:
That looks like a cleaner way of doing this for sure. I'll add it to the backlog for 2.0 just so we have a reference.
Thanks for sharing this!

closed 2022-02-03 by pilotnbr1