OpenHD/OpenHD #379: seek thermal support

hex0r, 2020-10-24:
Hi everyone,

I try to stream thermal imaging data using gstreamer without success

try to setup v4l2 loopback:
seek_viewer --camtype=seekpro --colormap=11 --mode=v4l2 --output=/dev/video0

then using the following gstreamer pipe:
gst-launch-1.0 v4l2src device=/dev/video0 ! video/x-raw,width=320,height=240,framerate=15/1 ! omxh264enc control-rate=1 target-bitrate=600000 interval-intraframes=1 periodicty-idr=1 ! h264parse config-interval=3 ! rtph264pay ! udpsink host=127.0.0.1 port=5600

get this error :
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data stream error.
Additional debug info:
gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming stopped, reason not-negotiated (-4)
Execution ended after 0:00:00.003602959
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...

please could anybody give me a hand 
thx

I have the following setup:
***  raspi4 
***  seek thermal pro

steveatinfincia, 2020-10-25:
Can’t use omxh264enc on buster, use the seek pipelines in the default config they have v4l2h264enc instead.

There’s still something not quite right for seek support at the moment, probably an argument to seek_viewer not being passed correctly, but you can test manually like you’re doing if you like. 

I have a seek compact pro coming in the next few days, I’ll get it fixed up :)

hex0r, 2020-10-25:
thx ,man

hex0r, 2020-10-27:
hi,
First of all openhd really nice project but i am very new to it. 
So sorry for stupid question
As I understand the code right, the rsavid output is send through tx_rawsock to ground
and USB cams are forwarded udpsource on 5600 and sends it via wifibroadcast
please correct me if this is wrong

So,
I download the v202 (Buster) prebuild image, startup everything works perfect with raspicam.
Now i followed the guide to use USB cam 
create air.txt
SecondaryCamera=USB
IsCamera1Enabled=1
USBCamera="gst-launch-1.0 videotestsrc ! video/x-raw,width=160,height=120,framerate=30/1 ! videoconvert ! v4l2h264enc  ! h264parse config-interval=3 ! rtph264pay ! udpsink host=127.0.0.1 port=5600"

startup again, now video at all 
what could be the reason for that?
my setup:
   Ground:
       Raspi3+
       2*WN722N
   Air:
       Raspi4
       1* AWUS036NHA

steveatinfincia, 2020-12-03:
This should be all fixed and working well now, tested by a lot of people. Let us know if you still have trouble.

R2AGQ, 2022-07-27:
> This should be all fixed and working well now, tested by a lot of people. Let us know if you still have trouble.
Hello, help pls)

Seek thermal compact pro
Version: 2.0.12B and 2.0.8 not work, error identical
Tried other raspberries, flash.... error identical

# >>>>>>>>>>[OPTIONAL SEEK-USB CAMERA DRIVER SETUP]--
# Values: Y N
LoadSeekDriver=Y
# You should change this in USBCamera as well if you need to change it at all
SeekFramerate=9
# This affects the colorization used for the thermal range, experiement to find the one you like
SeekColormap=1
SeekRotate=0
# Can also be seekpro
SeekModel=seekpro
________________
USBCamera="gst-launch-1.0 v4l2src device=/dev/video0 ! video/x-raw ! videoconvert ! v4l2h264enc  ! h264parse config-interval=3 ! rtph264pay ! udpsink host=127.0.0.1 port=5600"
_________________
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data stream error.
Additional debug info:
gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming stopped, reason not-negotiated (-4)
ERROR: pipeline doesn't want to preroll.
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...

raphaelscholle, 2023-04-15:
currently we only support thermal cameras which expose themselfes via uvc, these cameras are autodetected and should work

closed 2023-04-15 by raphaelscholle