OpenHD/OpenHD #380: Are there any limits in term of quality?

bkzspam, 2020-11-05:
Hello,
I will buy Raspberry Pi 4 for use in this project. I don't have any issue with latency except it's more than 250ms. I am not sure if there is a loss of quality, I am going to use Canon 7D Mark ii with HDMI connection. So is quality will be decreased? And are there any bandwidth limit?
Thanks

steveatinfincia, 2020-12-03:
Quality depends on the settings (bitrate mostly), but yes some cameras with HDMI output have added latency compared to a CSI camera like the picamv2.

Video bitrate is limited by the radio settings, but you can increase both if you want.

RespawnDespair, 2020-12-04:
Like Stephen said, there are limitations on the maximum bandwidth available, and the tradeoff is always: **more bandwidth equals shorter range**.
That said, you should be able to get good picture quality in 720p or 1080p with the appropriate bandwidth, but 720p is what is mostly used. Range on 1080p will be limited. 
In the pipeline you mentioned, using the Canon 7D and the HDMI to CSI board, most latency will be introduced by the camera and the adapter board. The latency added by Open.HD and the transmission of the signal will be less of a factor.
I will close this issue as i believe your question is answered, if you have any issues, feel free to reopen this issue or create an new one. (And let us know how it went, i'm curious about the picture quality of a DSLR through Open.HD).

closed 2020-12-04 by RespawnDespair