OpenHD/OpenHD #382: IP camera support as single camera

schs2, 2020-11-09:
I made config changer fro wiki -
https://github.com/OpenHD/Open.HD/wiki/Software-~-Advanced-~-USB
After reboot video showed only in PIP window. Fullscreen video absent.

I made small patch:
1) add new parametr in config IPCameraOnly=Y
2) patch script RxForwardSecondaryRTPAndDisplayLocally.sh 
------
#!/bin/bash

source /usr/local/share/wifibroadcast-scripts/global_functions.sh

cd /usr/local/share/cameracontrol/IPCamera/svpcom_wifibroadcast/

if [ -e "/tmp/settings.sh" ]; then
    OK=`bash -n /tmp/settings.sh`
    if [ "$?" == "0" ]; then
        source /tmp/settings.sh
    else
        echo "ERROR: wifobroadcast config file contains syntax error(s)!"
        collect_errorlog
        sleep 365d
    fi else
    echo "ERROR: wifobroadcast config file not found!"
    collect_errorlog
    sleep 365d
fi

detect_os

migration_helper

NICS_LIST=`ls /sys/class/net/ | nice grep -v eth0 | nice grep -v lo | nice grep -v usb | nice grep -v intwifi | nice grep -v wlan | nice grep -v relay | nice grep -v wifihotspot`

/usr/local/share/cameracontrol/IPCamera/svpcom_wifibroadcast/wfb_rx -u 5612 -p 23 -c 127.0.0.1 -n $VIDEO_BLOCKS_SECONDARY -k $VIDEO_FECS_SECONDARY -K /tmp/rx.key $NICS_LIST >/dev/null 2>/dev/null &

if [ "$IPCameraOnly" == "Y" ]; then
    gst-launch-1.0 udpsrc port=5612 !  tee name=t !  queue ! udpsink host=127.0.0.1 port=5620 t. ! "application/x-rtp,media=video" ! rtph264depay  ! video/x-h264, stream-format="byte-stream" ! filesink location=/dev/stdout | /usr/local/bin/hello_video.bin.240-befi ${HELLO_VIDEO_ARGS}
else
  if [ "$ENABLE_QOPENHD" == "Y" ]; then
    gst-launch-1.0 udpsrc port=5612 !  tee name=t !  queue ! udpsink host=127.0.0.1 port=5621 t. ! udpsink host=127.0.0.1 port=$VIDEO_UDP_PORT2
  else
    gst-launch-1.0 udpsrc port=5612 !  tee name=t !  queue ! udpsink host=127.0.0.1 port=5621 t. ! "application/x-rtp,media=video" ! rtph264depay  ! video/x-h264, stream-format="byte-stream" ! filesink location=/dev/stdout | /usr/local/bin/hello_video.bin.240-befi ${HELLO_VIDEO_ARGS}
  fi
fi
----

steveatinfincia, 2020-12-03:
This should already be working now in 2.0.6, no changes needed. I and a lot of other people have been testing it recently.

closed 2020-12-03 by steveatinfincia

schs2, 2020-12-04:
Thanks for fix!