OpenHD/OpenHD #384: Reboot AirPi with HDMI-CSI Setup fails

FranzPVal, 2020-11-18:
I have trouble rebooting the AirPI from my GroundPI (QOpenHD).

**My Setup**
AirPi: Pi Zero / Pi 3A+ combined with a TC358743XBG CSI-2 adapter and a Hawkeye Firefly 4K Split attached to it.

[CSI-2 Adapter iI used (AliExress)](https://www.aliexpress.com/item/4000169281286.html)

The Firefly was on 720p60 HDMI output (force HDMI on / off doesn't matter).

GroundPi: Pi 4B

Both sides running v2.0.4 buster. The problem was also present on 2.0.2 stretch on AirPI / 2.0.2 buster on ground.

**What happen**
After I click on reboot the AirPI reboots and "came back" after a while. But: There is no video stream. The OSD shows 0.0MBps and the CPU load indicates that no encoding is happen.

The same behavior I can reproduce with a power cycle of the camera alone or by disconnecting-reconnecting the HDMI cable. The stream will not recover.

I got an image every time I power cycle the AirPI (with the CSI-2 board attached), even if I left the Firefly powered.

**To Reproduce**
Steps to reproduce it:
1. Start Air + Ground PI
2. After both are operate nomally, got in the QopenHD power menu and click + confirm "restart AirPi"
3. Wait till messages indicates the reboot
4. Now the screen are black, telemetry working but no video. OSD shows 0.0MBps

**Expected behavior**
After a reboot or a camera power cycle or a HDMI re-plug the AirPI should start stream encoding (again).

**Configuration**
I tried the default openhd-settings-1.txt except FPS (59.9) / FREQ and EXTRAPARAMS parameter.

I tried EXTRAPARAMS="-cd H264 -n -fl -ih -pf high -md 1 -awb off -awbg 1.0,1.0 -ex off" from the [wiki HDMI in cards](https://github.com/OpenHD/Open.HD/wiki/AddOns-~-HDMI-in-Cards) and the suggested / commented one beside the default.

steveatinfincia, 2020-12-03:
Yea this is a weird one, it's likely caused by raspivid and the older firmware based driver for that toshiba chip, I suspect the newer camera stack in Raspbian won't have the same problem.

FranzPVal, 2020-12-04:
Do you have changed to the newer camera stack already in 2.0.6 or it is planned for 2.1?
I like to repeat the test with this version.

FranzPVal, 2020-12-08:
Problem still exist in 2.0.6.

steveatinfincia, 2020-12-08:
We’ll change it in 2.1 yes

raphaelscholle, 2023-01-31:
should be fixed in 2.3

closed 2023-01-31 by raphaelscholle