OpenHD/OpenHD #385: [Feature Request] Simplified multi model (AirPi) settings

FranzPVal, 2020-11-19:
**Current situation**
We modify openHD-settings-1.txt or select a profile and use SmartSync to keep the parameters matched between GroundPi and AirPi. All parameters can be changed only with a reboot.

**Suggested Improvement**
An AirPi should broadcast his video stream settings (blocksize, blocks, fec-blocks, ...) at leased once per 5 seconds like telemetry. The GroundPi waits after booting for this information and compare it every time there is no stream received (for a while). On an GroundPi there should only FREQ and BANDWIDTH matters.

For convenience QOpenHD should offer a (drop-down) menu for selecting FREQ (filtered for usable only) and BANDWIDTH (if hardware supports it) that results in an immediately change to the selected channel / frequency.

steveatinfincia, 2020-11-19:
This is how 2.1 works :)

RespawnDespair, 2020-12-04:
To expand a little on the above answer, yes, we share your opinion. While it is currently possible to use Open.HD with multiple models it is not the most apparent or user friendly experience. The re-architecture of 2.1 addresses this issue at a software level, but even then we need to make sure the QOpen.HD supports this as well.

Our main focus now is getting 2.0 to a point that is stable enough for people to fly with so we can buckle down and get the 2.1 version to a release candidate stage so people can start testing that in the field.

Sadly this means we probably won't support this user friendly functionality in the current 2.0, but it will have to wait to be implemented in QOpen.HD along with the 2.0 release.

I will leave this ticket open and attach the appropriate label.

FranzPVal, 2020-12-04:
I'm aware of the fact, that such things won't and shouldn't implement in V2.0.x anymore. As a developer for embedded µC I understand this.
So keep V2.0.x stable (bug fixing) but I like to test V2.1 as early as possible (but I have only a standard / limited setup now: no USB, IR or IP cams).

RespawnDespair, 2020-12-04:
That's why I left the ticket open, as soon as we feel comfortable sharing 2.1 with other devs/users you will be notified immediately.
We can use all the help, and standard setups are what gets tested first, so no worries.

closed 2022-10-16 by raphaelscholle