OpenHD/OpenHD #386: IP camera support as single camera in 2.0.4

schs2, 2020-11-19:
When I use alone IP camera on air side I got fullscreen video on PI groudstation and only PIP video on android phone.
No way to switch video on phone to fullscreen.
I make small patch for fix trouble:
Need to patch file RxForwardSecondaryRTPAndDisplayLocally.sh 
patch:
32c32
<     gst-launch-1.0 udpsrc port=5612 !  tee name=t !  queue ! udpsink host=127.0.0.1 port=5621 t. ! "application/x-rtp,media=video" ! rtph264depay  ! video/x-h264, stream-format="byte-stream" ! filesink location=/dev/stdout | /usr/local/bin/hello_video.bin.240-befi ${HELLO_VIDEO_ARGS}
---
>     gst-launch-1.0 udpsrc port=5612 !  tee name=t !  queue ! udpsink host=127.0.0.1 port=5620 t. ! "application/x-rtp,media=video" ! rtph264depay  ! video/x-h264, stream-format="byte-stream" ! filesink location=/dev/stdout | /usr/local/bin/hello_video.bin.240-befi ${HELLO_VIDEO_ARGS}

steveatinfincia, 2020-12-03:
This is ok for testing but you wouldn't want to change it like that all the time.

We're about to release new stuff that will do what you're trying to do :)

closed 2020-12-03 by steveatinfincia

schs2, 2020-12-04:
Thanks for fix!