OpenHD/OpenHD #387: Not an issue but a question

bjoern79de, 2020-11-20:
If I'm informed right, around 50% of the latency comes together on the pipeline from the camera (rolling shutter) over the gpu to raspivid. What do you think about the idea to offload that pipeline to an fpga unit by reading out a common global shutter cam over a broader interface, using H.264/H.265 HW encoding and passing the compressed stream over Ethernet/UDP to Open.HD?

I've found some open encoder approaches for FPGAs - no idea how "ready to test" they are on a FPGA-devboard.
https://github.com/bcattle/hardh264
https://github.com/Bearzeng/h.265_encoder

steveatinfincia, 2020-11-20:
Hey :)

Yes it's been discussed for a while, there are advantages and some pretty signifiant downsides but I'm up for it if the details could be worked out, and I have experience working on FPGAs as do a few others around the project.

The advantage would be the ability to use an encoder core and image processing core (debayer, noise reduction, etc) with very low latency, and there are both open source and closed IP for all of that around (closed isn't out of the question if it were practical to use, after all the GPU on a raspberry pi isn't open source either). 

For it to work you would need to connect CSI/LVDS or HDMI cameras, but there are ways to do that and some pre-made boards that have those kinds of connections already. I have a Pynq Z2 that has HDMI input right to the FPGA fabric, Zynqberry seems to as well, along with a CSI port.

The Pynq has a Zynq 7020 chip, which can easily fit a full h264 encoder and any necessary HDMI/CSI/ISP cores, but the ZynqBerry has a Zynq 7010 chip, and it's quite possible that the encoder alone would take up most of the resources (or at least make it difficult for the synth tools to meet timing).

If custom boards were created instead you could make whatever you wanted, but the cost might not be something anyone would be willing to pay compared to something that gets reasonably close and is widely available like the Jetson.

The one exception to that rule is that a lot of people value open hardware and open source software and prioritize it over cost, which precludes stuff like DJI from even being considered.

bjoern79de, 2020-11-20:
I have actually bought the DJI stuff a while ago and sent it instantly back because it interferes the analog guys and it was even impossible to align their "Custom OSD" against inav.

In my opinion the CM4 could be a game changer for this project because of it's form factor and it's stackability.
I'm currently trying to design a CM4-Hat (same form factor as the CM4 itself) with two AR9271 modules, amplifiers PSU and minimum needed connectors. Maybe it's possible to extend such a stack more or less easily by another layer containing a FPGA and it's base peripherals. Chinese custom pcb manufacturers like jlcpcb are not really expensive.

steveatinfincia, 2020-11-20:
> I have actually bought the DJI stuff a while ago and sent it instantly back because it interferes the analog guys and it was even impossible to align their "Custom OSD" against inav.

Yea there are a lot of issues with DJI, a lot of people around the project are never going to use their stuff again.

> In my opinion the CM4 could be a game changer for this project because of it's form factor and it's stackability.

The CM4 is a big deal yes, I wish they had put a real USB controller on it like the Pi4B has, instead it uses the older built-in USB OTG controller which has caused problems on all the pi models for a long time.

If you make a hat you might consider it, but I can imagine routing PCIe lanes complicates things a bit and would increase cost too.

> I'm currently trying to design a CM4-Hat (same form factor as the CM4 itself) with two AR9271 modules, amplifiers PSU and minimum needed connectors. Maybe it's possible to extend such a stack more [...]

Very nice!

RespawnDespair, 2020-11-25:
@bjoern79de A side project has started where we are designing 2 CM4 carriers. One for GND and one for AIR. This might not be your specific use case, but we could really use someone with EE experience to help with the design. Would you be open to working on that with us?

bjoern79de, 2020-11-25:
My EE experience is quite "hobbyistic", but yes!

RespawnDespair, 2020-11-25:
That's the qualification for most of the team members, 'hobbyistic'. Are you in the Telegram group as well? If so, could you contact me via DM there? I'm 'Jelle' in the Telegram group.

steveatinfincia, 2020-11-27:
I reached out to a2e about their h264 encoder core, which they say will fit on a Zynq 7010 like the ZynqBerry and ZynqBerryZero. 

Those boards are cheap enough to be within reason compared to DJI stuff, and they have both a CSI and HDMI connector usable for input right on the board.

I won't have time to mess with fpga stuff for a while but if it's possible to get what we need to fit on those boards it would be worth it and I could pay the upfront cost for it.

bjoern79de, 2020-11-27:
Sounds good!
Since the Zynq has included ARM-Cores: Do you think about running the whole
Open.HD-Stack onto it?
Have you already asked a2e about the terms? I guess most of these
IP-vendors are working exclusively together with "industrial customers"...

steveatinfincia, 2020-11-27:
> Sounds good! Since the Zynq has included ARM-Cores: Do you think about running the whole Open.HD-Stack onto it? Have you already asked a2e about the terms? I guess most of these IP-vendors are working exclusively together with "industrial customers"... 

Yep the transmission stuff and the rest of OpenHD could run on the ARM side easily, we'd basically be making a custom SoC.

a2e seems like a lost cause, they say their h264 core won't fit on a Zynq 7010 and the terms were absurd anyway. It may have fit at one time, I saw someone talking about it in the past.

However I have a 7020 with HDMI input here I can use for proof-of-concept when I get a chance, h264 will fit on one of those easily. If it works out well perhaps we can think about smaller hardware.

bjoern79de, 2020-11-28:
I see. Another interesting starting point could be HDMI2USB. It's based on
Litex. But it embeds a mpeg encoder (too slow?).

closed 2022-10-17 by raphaelscholle