OpenHD/OpenHD #388: RTL8812AU reported signal strength

FranzPVal, 2020-11-23:
There is a strange behavior I noticed while testing with a RTL8812AU adapter as open.HD receiver.

If I connect only one antenna to the the adapter and a dummy load (or let it open) th the other SMA-RP connector the reported dBm differs exceptional!

* antenna connected to the main connector (is used for TX on non STBC modes) results in showing -82dBm
* antenna connected to the secondary connector shows -56dBm
* two antennas connected = average of both values ~ 70dBm is displayed

This behavior is the same even with MCS+STBC!

Reception sensitivity is about the same ("I got an image") - so antenna diversity seems to work.

**Test setup**
* AirPI zero with PI V1 cam, TX power set to 1 (-10dBm measured) on a 30dB attenuator + omni circular antenna
* Receiving Ground PI 4 using a 10dB attenuator  and a omni antenna / repeated with -22dB and patch antennas

RespawnDespair, 2020-12-04:
Thanks for the detailed report, by the best of my knowledge the dBm values are retrieved from the Linux driver itself. No real magic calculation takes place in Open.HD. It is very likely that my understanding is limited, but could you explain what you would have expected in the scenarios you described?

FranzPVal, 2020-12-04:
I guess that there is no additional calculation or averaging in open.HD for the OSD element. So you're right about that this is something the driver reports faulty or "too raw".
Expectation:
In all situations (one antenna on MAIN, one antenna on AUX, 2 antennas on both) there should report about the same signal strength because it is the same energy that is received. I have only 2 types of RTL87812AU (USB-Stick and Taobao) so my test capabilities are a bit limited... Please just repeat my test for confirmation.

Consti10, 2023-02-23:
We did a couple of improvements in this area in evo, please report again

closed 2023-02-23 by Consti10