OpenHD/OpenHD #390: Distance to home is 0

Justinas1980, 2020-11-26:
Sometimes (not always) Distance to Home is 0, everything else in OSD is shown normaly. One time this was with 2.0.4 version and today with 2.0.5, I tried reboot Air ant Ground Pi but nothing changes. RTH is working, so controller knows where „Home“ is. My AirPi is 3A+, Ground 3B, Filght controller - Pixhawk.

steveatinfincia, 2020-11-27:
Ok we’ll check it out

pilotnbr1, 2020-11-27:
What gps hdop numbers are you seeing?

Justinas1980, 2020-11-27:
> What gps hdop numbers are you seeing?

In a field I don't have mouse attached to Pi, but plane always see from 9 to 11 sattelites. I've tried arm (with button) after everything is booted and arm first, then connecting ground Pi - with same results. 
If weather will be flyable will check home coordinats tommorow.
[image]

pilotnbr1, 2020-11-27:
In your screenshot your hdop looks good at .81. If memory serves me
correctly, if it goes above 2.5 you might encounter problems with home
being set.

user1321, 2020-12-03:
Same thing happened to me last year with old system. Reason was in demaged file system and  some module was unable to compile itself as result always 0 home distance. Maybe you have same situation.
Just try to write image to SD card again. (Probably ground unit if I remember correctly )

Btw, guys, did you find a way to protect SD card? 
(Except 50$ industrial sd cards ) Maybee that can be used somehow?
https://github.com/BertoldVdb/sdtool
 Or maybe if not protect, than add some tool that will scan each file in system and compare MD5 to find demaged files if there any.

steveatinfincia, 2020-12-03:
Hey :)

> Same thing happened to me last year with old system. Reason was in demaged file system and some module was unable to compile itself as result always 0 home distance. Maybe you have same situation.
> Just try to write image to SD card again. (Probably ground unit if I remember correctly )

We don't have anything compiling on the system anymore so it's probably something in code, there is some logic that should be looked at again.

> Btw, guys, did you find a way to protect SD card?
> (Except 50$ industrial sd cards ) Maybee that can be used somehow?

I use $7 Samsung Evo Select cards and never have a problem, I power off randomly all the time and everything is ok :D 

It really just seems to be specific cards that have issues, they suddenly lose power and who knows what was going on in the SD controller inside the card at the time, data loss city. That's generally what I see with the Sandisk Ultra cards, even when I was using them in custom pi security cameras a single power cycle could break the system and fail to boot again. Many many trips outside in freezing weather to retrieve and take them apart before I changed cards :D

RespawnDespair, 2020-12-04:
Adding this to the 2.0 stability effort.

closed 2023-01-31 by raphaelscholle