OpenHD/OpenHD #391: i want to use single usb

Wangzs0228, 2020-11-29:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Configuration and debug logs**
Please attach the relevant openhd-settings*1.txt and any debugging logs available

**Hardware:**
 - Air side
 -- Pi model
 -- Camera(s)
 -- Wifi card(s)
 -- Other hardware
 - Ground side
 -- Pi model
 -- Wifi card(s)
 -- Ground Control Station(s)
 -- Hotspots - do you connect you GCS using ethernet, wifi and/or usb tethering
 - RC/telemetry uplink
 -- Do you use RC over Open.HD, or a separate rf link?

**Client devices (in case of hotspot/tethering issue)**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - App - FPV_VR, QGC, and version
 - Version [e.g. 22]
 - Connection method

**Additional context**
Add any other context about the problem here.

>

steveatinfincia, 2020-12-03:
You'll probably have to use one of the other example USBCamera lines, the one that mentions uvch264src is for a specific kind of camera (Logitech C920 for example), most USB cameras that support h264 don't work quite the same way.

What camera is this, and what are you using to test it on other platforms?

I can write you a USBCamera line for it once I know what it is.

Wangzs0228, 2020-12-04:
> You'll probably have to use one of the other example USBCamera lines, the one that mentions uvch264src is for a specific kind of camera (Logitech C920 for example), most USB cameras that support [...]

closed 2020-12-04 by Wangzs0228