OpenHD/OpenHD #392: Openhd doesn't work on CM4,it can't boot

wuzz2zy, 2020-12-02:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Configuration and debug logs**
Please attach the relevant openhd-settings*1.txt and any debugging logs available

**Hardware:**
 - Air side
 -- Pi model
 -- Camera(s)
 -- Wifi card(s)
 -- Other hardware
 - Ground side
 -- Pi model
 -- Wifi card(s)
 -- Ground Control Station(s)
 -- Hotspots - do you connect you GCS using ethernet, wifi and/or usb tethering
 - RC/telemetry uplink
 -- Do you use RC over Open.HD, or a separate rf link?

**Client devices (in case of hotspot/tethering issue)**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - App - FPV_VR, QGC, and version
 - Version [e.g. 22]
 - Connection method

**Additional context**
Add any other context about the problem here.

wuzz2zy, 2020-12-02:
I want to use a CM4 with emmmc,but it doesn't work

steveatinfincia, 2020-12-03:
Needs an updated kernel and gpu firmware I'm sure, we'll update it soon.

RespawnDespair, 2020-12-04:
CM4 with eMMC is also on the list of several other members of the project. Are you intending to develop a custom carrier? A project to do just that is already underway, so maybe we can exchange notes. Meanwhile i will add some labels to this issue to keep you updated.

closed 2022-10-16 by raphaelscholle