OpenHD/OpenHD #393: Possible with virtualbox and raspberry img ?

Ek91, 2020-12-02:
Is this possible to to with virtualbox and raspberry desktop img ?

steveatinfincia, 2020-12-03:
You mean run the entire image? It wouldn't be worth it to use the pi image that way, a lot of things like the hardware decoder/renderer are specifically for the pi.

We will have x86 images soon though that should work as-is.

RespawnDespair, 2020-12-04:
Running the Open.HD image in any kind of emulator is going to be tricky. Like Stephen said it depends heavily on the actual hardware of the Raspberry Pi.

If your usecase is for development, consider getting a real Raspberry Pi and connect to it through Ethernet from your desktop. This is the way a lot of us do development.

If you are looking for a cheap way to test the system, this unfortunately is not the way to go.

In the future we are supporting more platforms aside from the Raspberry Pi, one of those platforms will be X86, which allows you to run the system on a  Linux desktop, be aware there is still a requirement for the [supported ](https://openhd.gitbook.io/open-hd/hardware/wifi-adapters)WiFi cards even in that scenario.

Hopefully this answered your question, if not feel free to reopen this issue or create a new one.

closed 2020-12-04 by RespawnDespair