OpenHD/OpenHD #394: Ground battery low warning message in OSD

macrokernel, 2020-12-02:
It would be nice to be warned when the ground battery voltage is getting too low. The voltage threshold should be user-configurable, as some may use Li-Po or other kinds of batteries which can be discharged down to 2.5 Volts or lower, but make it default to Li-Ion as these are more wide-spread. Of course this feature can only work if the ground station has a voltage sensor.

steveatinfincia, 2020-12-03:
Very good idea, there is an OSD widget for ground battery and power right now too but it isn't enabled at the moment, needs some thinking about how to present the info since there are multiple batteries

macrokernel, 2020-12-04:
Some ideas on multiple batteries.

Perhaps allow adding multiple battery widgets to the OSD and let user select a voltage source to be displayed in a widget from the widget configuration menu in the OSD.

Or maybe add battery voltages to the existing ground/air CPU usage and Temperature widget. And make battery display configurable: percentage, batt voltage, cell voltage.

RespawnDespair, 2020-12-04:
I am working on the ground power widget. It is almost complete. I will check with you if it meets your expectations.

macrokernel, 2020-12-04:
Fantastic! Thank you Jelle.

closed 2023-01-31 by raphaelscholle