OpenHD/OpenHD #395: Single camera issue

Wangzs0228, 2020-12-04:
I am the one who mentioned the USB camera problem(I don't know why the previous question disappeared).
I have found that the 'uvch264src'  isn't suitable for the camera I am using. So I change the code in openhd-setting-1.txt 
from
USBCamera="gst-launch-1.0 uvch264src device=/dev/video0 initial-bitrate=3000000 average-bitrate=3000000 iframe-period=1000 name=src auto-start=true src.vidsrc ! queue ! video/x-h264,width=1280,height=720,framerate=30/1 ! h264parse ! rtph264pay ! udpsink host=127.0.0.1 port=5600"
to
USBCamera="gst-launch-1.0 v4l2src device=/dev/video6 ! video/x-h264,width=1280,height=720,framerate=30/1 ! h264parse ! rtph264pay ! udpsink host=127.0.0.1 port=5600"
and it works. However, the latency approached to 500 ms. I want to know, is it normal phenomenon? or what should i do if I want to shorten it. Thank you!

steveatinfincia, 2020-12-04:
On that specific camera it may be normal, what is it?

Wangzs0228, 2020-12-04:
when using the codes i have changed, openhd works as expected. But the latency approaches 500 ms. My camera is a traditional AV camera and I used the adapter board to convert it to USB H264 camera

Wangzs0228, 2020-12-04:
> On that specific camera it may be normal, what is it?
> […](#)

> On that specific camera it may be normal, what is it?
> […](#)

when using the codes i have changed, openhd works as expected. But the latency approaches 500 ms. My camera is a traditional AV camera and I used the adapter board to convert it to USB H264 camera

RespawnDespair, 2020-12-04:
Can you be more specific about which camera and which adapter board? Is it an HDMI to CSI board, or something else?

Wangzs0228, 2020-12-04:
> Can you be more specific about which camera and which adapter board? Is it an HDMI to CSI board, or something else?
camera:  https://item.taobao.com/item.htm?spm=a1z09.2.0.0.5c002e8dppyfZi&id=600195385881&_u=42eja57h9682
adapter board：https://item.taobao.com/item.htm?spm=a1z09.2.0.0.5c002e8dppyfZi&id=559258887574&_u=42eja57he400

RespawnDespair, 2020-12-04:
That looks interesting, I don't think anyone has experience with this board.
If you use the board to directly view the video on a Linux computer, what is the delay then?

Wangzs0228, 2020-12-04:
> That looks interesting, I don't think anyone has experience with this board.
> If you use the board to directly view the video on a Linux computer, what is the delay then?

About 100 ms or less,  I didn't measure it carefully

RespawnDespair, 2020-12-04:
Unless the gstreamer pipeline is completely wrong it shouldn't add 400ms.

What is the pipeline you use to directly test on Linux? Without Open.HD in between?

Wangzs0228, 2020-12-04:
> completely

 gst-launch-1.0 v4l2src device=/dev/video0 ! video/x-h264,width=1280,height=720,framerate=30/1 ! h264parse ! avdec_h264 ! videoconvert ! xvimagesink sync=false

i run this in my Linux, and the delay is less than 100 ms ,so i use it  to replace these code in openhd, and get 500ms...I also wander why?

Wangzs0228, 2020-12-04:
> Unless the gstreamer pipeline is completely wrong it shouldn't add 400ms.
> 
> What is the pipeline you use to directly test on Linux? Without Open.HD in between?

Can you give me some advice about how to reduce the latency or how to check what's wrong with it?

RespawnDespair, 2020-12-04:
Just a quick follow-up, not sure this has anything to do with your issues, but what are you using in the AIR unit? Which Raspberry?

Wangzs0228, 2020-12-04:
> Unless the gstreamer pipeline is completely wrong it shouldn't add 400ms.
> 
> What is the pipeline you use to directly test on Linux? Without Open.HD in between?

Can you give me some advice about how to reduce the latency or how to check what's wrong with it?

> Just a quick follow-up, not sure this has anything to do with your issues, but what are you using in the AIR unit? Which Raspberry?

3b

RespawnDespair, 2020-12-04:
> Can you give me some advice about how to reduce the latency or how to check what's wrong with it?
Unfortunately no, maybe some of the resident wizards might spot a possible optimization, but I don't see it.

The 3B should be powerful enough to not introduce that amount of latency.
I hope some of the other members can read along and hopefully come up with good ideas.

We might need to order one of these boards to do some testing ourselves.

steveatinfincia, 2020-12-04:
If you're seeing a bunch of latency it's either something about the stream itself (some h264 cameras cause the decoder to do weird things, including extra latency), or the h264 encoder bitrate is a little bit high. If that adapter provides control over the h264 bitrate via v4l2 it would be possible to adjust it, but a lot of them don't.

I'll look and see if I can find one of those adapters to order,

Wangzs0228, 2020-12-07:
> If you're seeing a bunch of latency it's either something about the stream itself (some h264 cameras cause the decoder to do weird things, including extra latency), or the h264 encoder bitrate is a little bit high. If that adapter provides control over the h264 bitrate via v4l2 it would be possible to adjust it, but a lot of them don't.
> 
> I'll look and see if I can find one of those adapters to order.
> 
> The easiest way to test if it's the stream itself would be to connect a phone to the openhd hotspot and use fpv_vr, qopenhd, or qgroundcontrol and see if there's still latency.

Can I configure the network card into ad-Hoc mode and send the collected H264 data directly through the network card? Can it reach the same distance as Openhd?

air: gst-launch-1.0 v4l2src device=/dev/video6 ! video/x-h264,width=1280,height=720,framerate=30/1 ! h264parse ! rtph264pay ! udpsink host=10.0.0.1 port=5600
ground: gst-launch-1.0 udpsrc port=5600 ! "application/x-rtp, payload=96" ! rtph264depay ! decodebin ! autovideosink sync=false

Using the raspberry PI as the air side, the PC  as ground side, the above code delay is around 110~120ms, but I haven't measured the distance yet.

closed 2023-01-31 by raphaelscholle