OpenHD/OpenHD #397: Adding on AIR an rs232 gps

domenic64, 2020-12-04:
Is it possible to implement a rs232 gps on air?
i don't have a drone i would like to use openhd on a hidrofoil, soon I will do a test in the water to see how far I can get. otherwise I cannot have any gps telemetry data.
Thanks

domenic64, 2020-12-04:
[image]
[image]

domenic64, 2020-12-04:
water tests
[image]
[image]
first test, water entered

RespawnDespair, 2020-12-04:
Open.HD expects MAVLink messages on the serial port, so it is not possible to send raw NMEA data.
I have seen other users translate data to MAVLink on an Arduino. See #331 for example.

It might actually be easier to install a small F4 based Ardupilot Flight Controller. Even if you don't use it for anything else than the GPS. These boards will also allow Ardurover which might actually be useful on your hydrofoil?

domenic64, 2020-12-04:
There is no space in the box.

domenic64, 2020-12-04:
[image]

RespawnDespair, 2020-12-04:
I think Open.HD is not the ideal candidate for what you are trying to achieve. 
Some of the other projects out there such as [SVPCOM ](https://github.com/svpcom/wifibroadcast) allow for transparent UDP links between air and ground. In the future Open.HD will likely also support this, but at present we do not.

If you want to move forward with your project you might be better off trying to use that.

@steveatinfincia I am not wrong about the MAVLink requirement am I? Before I send someone packing...

steveatinfincia, 2020-12-04:
In 2.0 no, but next version If you connect a GPS to the air side and write a tiny bit of code you could generate the gps mavlink message and send it along over the air just like it was coming from a flight controller.

There are some other things that could benefit from having a simple gps connected too, so it may be worth supporting anyway.

happytm, 2021-08-23:
@steveatinfincia  Is this feature (gps mavlink message) implemented yet ?

Thanks.

closed 2022-10-16 by raphaelscholle