OpenHD/OpenHD #398: Ground video record from single IP camera

schs2, 2020-12-04:
Open.HD 2.0.4
I want to record on ground PI.
I set in config:
VIDEO_TMP=memory
VIDEO_FS=ext4
VIDEO_SAVE_FORMAT=mp4

After fly I insert usb stick, got messages about start writing and end writing.
I remove stick and connect to windows PC. No any video on stick, only telemetry and other info.

Can be add support video record from single and alone IP camera?
Best way - choose videosource for record.

RespawnDespair, 2020-12-04:
I'm not 100% sure, but I think the system doesn't record the video if coming from an IP camera now, even if having it as the only source. I've put it on the stability issues for 2.0.

I hope to get back to you soon.

schs2, 2020-12-04:
Video record start in script rx_functions.sh:

`            ionice -c 1 -n 3 /usr/local/bin/rx -p 0 -d 1 -b $VIDEO_BLOCKS -r $VIDEO_FECS -f $VIDEO_BLOCKLENGTH $NICS | ionice -c 1 -n 4 nice -n -10 tee >(ionice -c 1 -n 4 nice -n -10 /usr/local/bin/ftee /var/run/openhd/videofifo2 > /dev/null 2>&1) >(ionice -c 1 nice -n -10 /usr/local/bin/ftee /var/run/openhd/videofifo4 > /dev/null 2>&1) >(ionice -c 3 nice /usr/local/bin/ftee /var/run/openhd/videofifo3 > /dev/null 2>&1) | ionice -c 1 -n 4 nice -n -10 /usr/local/bin/ftee /var/run/openhd/videofifo1 > /dev/null 2>&1
`
but stream from IP camera get another script RxForwardSecondaryRTPAndDisplayLocally.sh 
May be add ftee to this script?

steveatinfincia, 2020-12-23:
The older code was never wired up to record anything but pi cams, likely because the "recording" system was very simple, but the code for OpenHD 3 will allow recording any camera to any storage location on air or ground.

closed 2023-01-31 by raphaelscholle