OpenHD/OpenHD #400: Support for NanoPi Duo2

howels, 2020-12-05:
The NanoPi Duo2's compact size and faster quad-core CPU make it a compelling choice for users looking for a small air-pi with more CPU than the Pi Zero.

The board features a camera connector and u.fl wifi antenna.

https://wiki.friendlyarm.com/wiki/index.php/NanoPi_Duo2

I have a board and am happy to test.

steveatinfincia, 2020-12-07:
We'll support those very soon (within 2 weeks), however I could caution that the onboard wifi is likely a Broadcom chip which technically can do broadcast transmission but last I checked it wasn't fast enough (same as the Pi), however I'll look again.

The camera connector may work, we'll have to try it. I expect for most people the nanopi boards will be primarily useful for usb h264 and IP cameras since for those it doesn't matter what board you use as long as it has USB and/or ethernet.

howels, 2021-01-13:
Any update on this? Thanks.

howels, 2021-02-18:
Any news on this device?

closed 2022-10-16 by raphaelscholle