OpenHD/OpenHD #402: OpenHD 2.1 Progress Tracking

steveatinfincia, 2020-12-13:
This issue exists to track progress on OpenHD 2.1 (which may end up being called 3.0 given that it's a near total rewrite).

There are a handful of things that aren't merged yet (video service), and some little things in existing code that still need to be done, but nothing major. 

Settings load is fairly simple and is what I'm finishing up at the moment, sync just requires a few Mavlink messages over the services channel. QOpenHD also needs to be switched over to use those messages to load/save settings but that should be done once 2.1 betas are released in ~2 weeks, so the ground/air parts are already implemented.

- [ ] openhd-interface (partially merged)
    - [ ] WiFi
        - [X] Streams
        - [X] Hotspot
        - [ ] Client (just requires running one of the dhcp clients automatically)
        - [X] Settings
        - [ ] Sync
    - [ ] Ethernet
        - [X] Hotspot
        - [ ] Client (just requires running one of the dhcp clients automatically)
        - [X] Settings
        - [ ] Sync
    - [ ] LTE/VPN (was originally based on just setting up the interface and running the providers background service)
        - [ ] Settings
        - [ ] Sync
- [X] openhd-status (merged)
    - [X] Log messages locally for dev/testing
    - [X] Log messages to file (stored with video recordings/telemetry)
- [X] openhd-security (merged)
    - [X] Generate air/ground keys from use-supplied phrase
    - [ ] Use newer settings system (just a different path to the file)
- [X] openhd-system (merged)
    - [X] profile setup
        - [X] Unit ID loading/generation (identifies each ground station/drone uniquely)
        - [X] Air/Ground determination
    - [ ] Platform detection
        - [X] Raspberry Pi
           - [ ] Carrier detection (may not matter)
        - [X] Jetson
            - [ ] Carrier detection (may not matter)
        - [X] PC
        - [ ] Rockchip platforms
        - [ ] i.MX 6 platforms
    - [X] Camera auto-detection
        - [X] Raspberry pi CSI
        - [X] Jetson CSI
        - [X] Rockchip CSI
        - [X] USB UVC
        - [ ] IP cameras (not strictly necessary, manual entry of URL in camera.conf will work)
    - [X] WiFi auto-detection
        - [X] Card type / capabilities
        - [X] Band support (uses netlink kernel interface to query the card)
    - [X] Ethernet auto-detection
        - [X] RNDIS/iPhone detect (for tethering)
        - [ ] CDC-ETH (doesn't require special handling beyond a driver name in the manifest)
    - [ ] LTE stick autodetection (this is mostly covered by ethernet detection, they're often just CDC-Ethernet or RNDIS)
- [ ] openhd-rc (May be wise to just generate Mavlink messages and send them though the microservice channel)
    - [ ] Ground transmission
    - [ ] Air receive/generate protocols (reuse existing code)
    - [ ] Settings load
- [ ] openhd-video (not merged)
    - [ ] Camera auto-setup
        - [X] Raspberry Pi CSI
        - [X] Jetson CSI
        - [X] USB UVC (includes thermal at the moment)
        - [ ] Thermal camera direct streaming (merge the existing drivers so they all use the same palette/span code)
        - [X] IP
        - [ ] Settings load/store
        - [ ] Sync
    - [ ] Recording
        - [X] Record to file (USB/SD)
        - [ ] RC trigger (channel tracking is done, just needs a function to switch to next file and trigger start)
        - [ ] Arm trigger (same as above)
        - [ ] Settings load/store
- [X] openhd-router (merged)
    - [X] Mavlink
    - [ ] LTM (needs a parser, can be done after betas released)
    - [ ] MSPv2 (needs a parser, can be done after betas released)
- [X] openhd-microservice (merged)
    - [X] GPIO
    - [X] Power
    - [X] Status logging
    - [ ] Sensor (not critical to have done immediately)
- [ ] QOpenHD
    - [X] Raspberry Pi
    - [X] Jetson 
    - [ ] PC (only requires building QOpenHD package, possibly Qt)
    - [ ] Switch settings to service/parameter system (should be done after 2.1 betas are released so the ground/air side is already in place)

olliw42, 2020-12-23:
@steveatinfincia 
I'm not sure where to leave that comment, so apologies if it is misplaced here, feel free to direct me as needed.

first off, really fantastic work you guys are doing. 

as regards Mavlink, and the particular way it is used, I think there are some comments in place however.

* You may recall the effort, triggered by this https://github.com/mavlink/mavlink/pull/1537 in the mavlink repo, to create a place for external dialects to exist within the mavlink repo. There had been a dev call today and the response towards that idea had been overall very positive, and chances are high it will come (not tomorrow, but maybe in some few weeks time). The implication would be that the dialect integration/handling would change (and hopefully much simplify, that's at least the idea). (this should not be a very difficult change to accommodate)

* From the code my impression is that you also use mavlink as a means to establish a sort of inter process communication, specifically between some of the services. Accordingly, you give the services specific mavlink ids. I understand the intention, but that's rather a sort of misuse of mavlink. That is, if one thinks about integrating OpenHD into a general mavlink system this would not be what one should do, and may cause also troubles. I believe that you want to integrate well with mavlink systems, at least as medium term goal, and this thus may also imply the need of some changes. Given that your chosen mechanism is so much at the heart, this actually may need substantial work to disentangle.

* For some of the messages in your dialect, official messages in the mavlink standard exist, e.g. for camera control. I guess it would be useful to eventually convert to using them. Also, the mavlink standard provides a TUNNEL message, which could allow you to do much of what the other messages are for. 

So, what I'd like to bring across is that as regards mavlink I think some aspects in the current code would need some changes in order to become more "mavlink-ish". I'm not sure if it is realistic to put this up as a goal for v3.0, but I think you should put it up somewhere, and maybe you could evaluate if it is realistic for v3.0 (it may mean some additional delay, I know, but as said, I think you want to integrate well with mavlink systems).

I hope you find these comments useful. All the best, Olli.

(PS: I'm a self-proclaimed "linux-nut", and I would need to be enabled to do development for OpenHD, but I would be genuinely willing to help with the mavlink stuff).

RespawnDespair, 2020-12-24:
@olliw42 We were already changing the MAVLink generation based on your earlier issue. That is in place in the microservices repo now, but we are also moving that repo to be part of the main Open.HD repo.
You raise some interesting points and I would like tot talk to you to get things sorted and come up with a plan of attack.
Can you contact me (Jelle Tigchelaar) via DM on Telegram or do you prefer another method of communication?

olliw42, 2020-12-24:
I'm not on telegram. What about zoom, or skype?

We need to also find a date. I won't be available today for sure, and our plans for the next days are not yet fully clear. Better to discuss this again tomorrow early afternoon, if possible for you.

btw, the MAVlink generation should even more simplify with the new external dialect scheme in place, it in fact should become trivial. You simply could get your dialect to be placed in the mavlink repo, and just would have to do the normal mavgen as with standard dialects too. (all not yet functional, but it will come rather in weeks than months).
(you may have seen hamishwillee's latest post: https://github.com/mavlink/mavlink/pull/1537#issuecomment-750721527)

RespawnDespair, 2020-12-24:
Google Meet is my platform of choice, but I was talking more about an inital 'slow chat' such as email or DM without polluting this issue :)
For me the coming days are also busy with Christmas stuff. I'll post another reply here to set something up in the days after Christmas.

olliw42, 2020-12-29:
@RespawnDespair 
sorry for the delayed response, was too busy with my Christmas present (3D printer) LOL
I made the proposed changes to the mavlink repo, see here https://github.com/mavlink/mavlink/pull/1537#issuecomment-752024889. It really works nicely, one just has to call the generator with the desired xml.

as regards the 'slow chat', this appears to be more difficult than one would think ... I was about to suggest using messages in https://discuss.openhdfpv.com/ ... but it seems messages are disabled ?!?!?
so, I guess, we simply could open a separate issue in this gitrepo here or open a thread in https://discuss.openhdfpv.com/ ... unless you think that for some reasons which would not be clear to me the discussion should be "secret", in that case we could go with email.

schs2, 2021-01-06:
When you plan to release 3.0? I got nanopi Neo3 for air side with IP camera and can test it.

olliw42, 2021-02-05:
as regards mavlink, the external/dialect mechanism has no been accepted by MAVLink

not sure what the plans are concerning mavlink, the OHD development seems to have stalled for a while now. I just want to say I'm still available and help if possible to mavlink-ize OHD :)

kasatka60, 2021-02-12:
> the OHD development seems to have stalled for a while now

Do you know what happened?

Alphaty, 2021-03-20:
So exciting that USB camera will be supported!

steveatinfincia, 2021-03-20:
They work automatically too, don’t need to set anything

ijustwant, 2021-04-16:
Hi. Any release plan for ver3 now ?

howels, 2021-04-20:
Some updates to indicate if this is stalled/short-of-maintainers/being-refactored/abandoned would be really helpful please.

pilotnbr1, 2021-09-10:
Currently image builder is not setting up the jetson image correctly. It is resulting in kernel modules failing to load (like csi cam)

rohitk1979, 2021-10-11:
Kindly look into the telemetry issues being faced with RTL8811 - if possible an option to merge telemetry with video packets can be provided in settings.
RTL8811 dongles are relatively inexpensive and smaller in size.

Thanks

MacDaddyFPV, 2021-12-10:
Currently removing stretch code from 2.1 repos. Also updating docker image tags and adding images to Github Container registry to prevent pull rate limit errors in the builds.

MacDaddyFPV, 2021-12-10:
As stretch was the only platform using OpenVG and Wifibroadcast-osd I suggest we remove these both from 2.1 to clean up the repo a bit more. I will create a branch with all stretch and relevant add-ons removed.

jpmarotta, 2022-02-11:
If you're looking for Jetson testers, I'm willing to dedicate a Nano 16Gb production module to the cause.  Would be interested in using it as an air unit with the HDMI adapter.

raphaelscholle, 2022-07-02:
Since we started a new "2.1" I've decided to close this milestone

closed 2022-07-02 by raphaelscholle