OpenHD/OpenHD #406: Latency testing

Consti10, 2021-01-15:
Hello,
Here I am going to share my latency test result(s) since Telegram is not best suited for this

Consti10, 2021-01-15:
I did some latency testing (again after ~4 years on the rpi) and on the jetson.
Pipeline is directly to phone via usb tethering, no wb in between
Full files / setup can be found here:
In short (end to end latency):
RPI at 720p 60fps has 83 - 87ms latency
Jetson nano at 720p 60fps has 62.5 - 70ms latency

https://drive.google.com/drive/folders/1ThlTyaDQLvHnDJRGqJVvsidg4HXEZPPe?usp=sharing

Consti10, 2021-01-15:
And when I enable GST_DEBUG for rpicamsrc it tells me that "some" latency is 43488us ==43.488ms

I am not sure "which" latency this is, but if end-to-end latency is 85ms
We can substract 17ms for decoding -> 68ms
And substract 44 ms -> 24
And we can probably substract ~16ms for the latency of the android display:
-> 8 ms unknown / transmission over network (though transmission over usb should not be more than a couple of microseconds)

Consti10, 2021-01-15:
Weirdly, when I go down to 640x480@60fps the value reported by rpicamsrc goes down to 14389us  (14.3ms)
So theoretically, I should see a decrease of the end-to-end latency of 28.608 ms
btw. an end-to-end latency of 56.4 ms
I am going to test

Consti10, 2021-01-15:
And yes, I can confirm that. The end-to-end latency at this resolution is between 50 and 66ms 
The decoding latency also decreased by ~2ms, so with deviation this probably adds up to the slightly better latency than expected )
So the question is: Why does the encoding time increase by 28ms when going from 640x480 to 1280x720 and how can we get rid of that increase / decrease it
It could be 
a) The encoder
b) Some weird binning from the ISP
Since I've validated via my end-to-end testing that the value reported by rpicamsrc seems to be accurate, I am going to test a couple more resolution / framerates
But only look at the value reported by gstreamer

Consti10, 2021-01-15:
These testings are a bit inconclusive:
1280x720 @49fps: ~25ms
1280x720 @30fps: ~38ms
1280x720 @60fps: ~45ms

For some reason, latency decreases by going from 30fps to 49fps but then greatly increases by going from 49fps to 60fps

Consti10, 2021-01-15:
~58ms @1080p30
The flakiness seems to be in direct corelation with the frequency of the h264 encoder core / isp frequency
But for some reason, on my rpi 3B+ it never goes above 300mhz even though the specs say it defaults to 400mhz
E.g. I sometimes get ~28ms at 720p60 but most often ~44ms
Not sure what's going on in the background. Using stock raspbian
But I can confirm, higher isp or h264 frequency -> lower latency ( as already stated 4 years ago :) ) but weirdly enough it actually decreases latency 
( weird because it the stream doesn't drop frames at 60fps - so at a lower clock speed the encoder somehow keeps up with the data coming in (eventually going for a higher latency path) but automatically enters a lower latency path if the frequency is high enough)

Consti10, 2021-01-15:
For my testings above I did not look at the gpu speed (using gpu as a synonym for h264_freq and isp_freq). That might explain the difference between 30 and 49fps

Consti10, 2021-01-17:
550mhz is the max I can get before it starts to get unstable.
Note that I did not increase the power for the chip / voltage or something

At fixed 550mhz I made the following observations:
720p60: ~20ms
720p30: ~33ms
1080p30: ~45ms

I think this shows that latency of encoder == frame time (since 720p30 has higher latency than 720p60 ) + extra value determined by clock speed in relation to resolution/framerate
E.g. the rpi encoder is only usable for 720p60 - for 1080p we would also need 60fps, but that's not possible
Or in general, latency for anything below 60fps is barely usable

Consti10, 2021-01-17:
Dang, I've just spent 1h adding the "slices" param from raspivid to gstreamear rpicamsrc just to find out that enabling slices greatly increases encoding latency at 720p60
e.g. it goes from 20ms up to 50ms (at least the value reported by gstreamer)
Important comment from 6by9:
"(Encoding a whole 1080P frame takes about 40ms, but is pipelined so that the next frame is started before the previous one completed, hence being able to achieve 30fps)"
From https://www.raspberrypi.org/forums/viewtopic.php?t=197504

Consti10, 2021-01-25:
https://www.raspberrypi.org/forums/viewtopic.php?t=197504

Consti10, 2021-02-07:
Some facts about resolution and bandwidth:
1280*720*120=110592000
1920*1080*60=124416000
-> 720p120 is lower bandwidth than 1080p60

YUV420 has 6 bytes per 4 pixels.
Aka 1.5 bytes per 1 pixel or  12 bits per pixel.
and according to
https://docs.baslerweb.com/pixel-format#bayer-formats
Bayer is either 8,10 or 12 bits per pixel.
So bayer is less or equal bandwidth than YUV420

SandroFTW, 2021-04-17:
> I did some latency testing (again after ~4 years on the rpi) and on the jetson.
> Pipeline is directly to phone via usb tethering, no wb in between
> Full files / setup can be found here:
> In short (end [...]

Awesome testing! I'd love to try out your settings myself, would you mind sharing them?

ATKCA, 2021-07-31:
I definitely agree the issue of latency needs to be a top priority. 

Good results Consti10

720p60: ~20ms
720p30: ~33ms
1080p30: ~45ms

These are all good results. Overclocking at 550Mhz can be made stable with more core voltage and or heatsinks if needs be.

I'm really interested in how consistent the latency is too. Would it be smooth enough for freestyle or racing quads? What are the minimum and maximum values?

Consti10, 2021-08-02:
> I definitely agree the issue of latency needs to be a top priority.
>
> Good results Consti10
>
> 720p60: ~20ms
> 720p30: ~33ms
> 1080p30: ~45ms
>
> These are all good results. Overclocking at 550Mhz can be made [...]

Well, this is the encoding latency, not end to end. it's missing isp & decoding & time to display. which is why I decided the rpi is not worth further investigations. I am focused on rv1126 now.
For comparison, it can do 1080p encoding in <10ms and 720p encoding in <7ms. SImilar seems to hold true for the isp.

MatthiasSchinzel, 2021-08-02:
This sounds great. Not familiar with the rv1126. Do you have a dev board?

Consti10, 2021-08-03:
yeah, if you are a developer you can join our telegram group.

white-rabbitGR, 2021-08-03:
> Well, this is the encoding latency, not end to end. it's missing isp & decoding & time to display. which is why I decided the rpi is not worth further investigations. I am focused on rv1126 now.
> For [...]

These are absolutely amazing latency results!
Did you manage to get OpenHD working on rv1126?

druckgott, 2021-08-19:
Where did this 44ms came from?
> And substract 44 ms -> 24

rsaxvc, 2021-11-02:
I looked at this some too - it seems there's a few things working against us on RPi:

1. MIPI CSI-2 bus is limited to 2 lanes. This puts a limit on how far we could double-framerate-and-decimate or for larger resolutions even limits the usable framerate.
2. The video input port only seems to trigger an IRQ when the full frame is transferred - average of 0.5 frame, max of 1 frame latency. I'm not sure if the VideoCore IV has a better mechanism for signalling.
3. VideoCore IV Encoder only seems to signal when the full batch of NALs is ready. Due to the encoder pipelining, this takes about 1.5 frames of latency for 1080p30, probably scales down with frame size. I saw some options to enable slice encoding, but these didn't seem to emit frames sooner though.
4. If the datalink goodput-bitrate is not a whole bunch faster than the video bitrate, transmission time adds up. Worst-case assume they're the nearly the same, it would add roughly 1 frame of latency to each NAL, but this goes down quickly as datalink-goodput-bitrate-to-video-bitrate ratio increases.

Consti10, 2023-03-18:
closing for now

closed 2023-03-18 by Consti10