OpenHD/OpenHD #409: Runcam webcam issues

Consti10, 2021-01-22:
The runcam webcam currently has a lot of issues. One of them is the fact that it can micro-freeze my PC running latest ubuntu.
However, if you have any news regarding this usb camera, here is the right place to share.

Consti10, 2021-01-22:
This pipeline results in an image, but it is played at ~1 fps with gstreamer complaining a lot.
Pipeline:
gst-launch-1.0 -v v4l2src device=/dev/video2 ! video/x-h264,width=1280,height=720,framerate=30/1,stream-format=byte-stream,profile=baseline ! avdec_h264 ! autovideosink

Error messages:
stbasesink.c(3003): gst_base_sink_is_too_late (): /GstPipeline:pipeline0/GstAutoVideoSink:autovideosink0/GstXvImageSink:autovideosink0-actual-sink-xvimage:
There may be a timestamping problem, or this computer is too slow.
WARNING: from element /GstPipeline:pipeline0/GstAutoVideoSink:autovideosink0/GstXvImageSink:autovideosink0-actual-sink-xvimage: A lot of buffers are being dropped

Consti10, 2021-01-22:
For some reason, the following pipeline gives a ~30fps experience:
gst-launch-1.0 -v v4l2src device=/dev/video2 ! video/x-h264,width=1280,height=720,framerate=30/1,stream-format=byte-stream,profile=baseline ! rtph264pay ! rtph264depay ! avdec_h264 ! autovideosink

Consti10, 2021-01-25:
So with my S9+ (exynos version) the decoding time for the runcam is ~10ms
This goes to show that it is possible in HW to decode the runcam stream with low latency, but >90% of decoders cannot do that
(and the decoder has to do some smart assumptions for that,too)
So I decided to measure the end-to-end latency without wb in between.
(transmission <1ms)
It gives me ~130ms end-to-end at 1080p30.
Subtract 10ms for decoding and 16ms for display:
= 104ms
This means the encoding delay (in a big sense) of the runcam is ~104ms
Now I assume that the issue with "1 nalu stuck" in the pipeline is true for the pipeline I used (since uvc gives a raw bit stream)
Then we can substract 33ms for the stuck frame, leaving 77ms for hw encoding + ISP processing on the runcam
That is slightly worse than rpi cam.

So for low latency, the runcam is just not usable

(And note that your smartphone decoder/ the rpi decoder probably has ~100ms decoding latency extra added on top)

Consti10, 2021-01-25:
Or in other words, the runcam is not usable for our project.

kasatka60, 2021-02-04:
Couldn't start with these parameters:
gst-launch-1.0 -v v4l2src device=/dev/video2 ! video/x-h264,width=1280,height=720,framerate=30/1,stream-format=byte-stream,profile=baseline ! rtph264pay ! rtph264depay ! avdec_h264 ! autovideosink.

Consti10, 2021-02-04:
1) check if the runcam is under /dev/video2 (might be dev/video0 if you are not using a laptop with integrated camera)
2) Re-connect the runcam, restart the pc. The runcam driver is buggy as hell to not usable on linux.

kasatka60, 2021-02-04:
Have you checked it on PC?
I tried to run it on raspberry pi openhd 2.0.8

Consti10, 2021-02-04:
I checked it on PC (ubuntu).

kasatka60, 2021-02-04:
ok.

kasatka60, 2021-02-07:
asus tinkerboard debian
gst-launch-1.0 -v v4l2src device=/dev/video4 ! video/x-h264,width=1280,height=720,framerate=30/1,stream-format=byte-stream,profile=baseline ! rtph264pay ! rtph264depay ! avdec_h264 ! autovideosink
latency ~150ms

openhd 2.0.8
USBCamera="usbreset 1d6c:0103; sleep 3; gst-launch-1.0 v4l2src device=/dev/video0 ! video/x-h264,width=640,height=480,framerate=30/1,stream-format=byte-stream,profile=baseline ! h264parse ! rtph264pay mtu=1024 ! udpsink host=127.0.0.1 port=5600"
latency ~450ms

Consti10, 2021-02-07:
The asus tinkerboard has a Mali™-T764 GPU. I strongly suspect the DE-coder of the tinkerboard is better in decoding the stream than the rpi. See my LiveVideo10ms repo and the discrepancy between decoders for specific h264 streams.

kasatka60, 2021-02-12:
Does your camera start normally on raspberry pi openhd?
To start the camera, I need to reconnect the camera power.

kasatka60, 2021-02-16:
In 1080p the video is freezing. In 720p, sometimes it freezes. In 640*480 there are practically no friezes at all. Do you have any tips?
I tried to change pi zero to pi3a+ and it's the same.

luqasz, 2021-06-05:
This issue has been reported to mailing list
https://sourceforge.net/p/linux-uvc/mailman/message/37226952/
https://sourceforge.net/p/linux-uvc/mailman/message/37067105/
Unfortunately without any answer. 
I've sent my report to mailing list.
I've tried some quirks, but no luck. Tried external powered USB hub. Camera works without any issues on mac.

Consti10, 2023-02-23:
Won't fix, don't buy this crap camera

closed 2023-02-23 by Consti10