OpenHD/OpenHD #410: RC on Ralink Cards

Yamaudi, 2021-01-24:
I was looking for a card under $15, works with RC, range not being priority.
I can't seem to find either rtl8812au or ar9271 cards that don't come all the way from China. I'm looking at the Ralink cards (3070, 5572, 3572, etc) people bought and used with varying levels of success. I'm planning to setup RC with INav using Zero AirPi and 4B GroundPi. I see people setting up video with ralink cards but I can't find anyone who used one for RC before. Checked the RC page and it says only 8812 and 9271 are supported but did that change recently by any chance?

Sorry if this is a naive question, I just found about OpenHD and not very informed about it yet.

Consti10, 2021-01-25:
I'd recommend buying from china, there is no reason not to

Yamaudi, 2021-02-13:
bought rtl8812au cards (bear extender turbo) from ebay for $16 a piece

closed 2021-02-13 by Yamaudi